wavelog/wavelog #689: Québec station location province breaks LoTW uploads

anthonydiiorio, 2024-08-03:
I cannot upload to LoTW with my Station Location Province set to Québec. I think LoTW does not like the accented é. 

In TQSL Quebec is selected with a regular e.

If I set my Province to Ontario for example I can upload my QSOs. I can also upload my QSOs if I set my station province to blank.

[image]

**Version Information**
- Server Software	Apache/2.4.61 (Debian)
- PHP Version	8.3.9
- MySQL Version	11.3.2-MariaDB-1:11.3.2+maria~ubu2204
- Wavelog 1.8 (Docker)

**Steps to Reproduce**
1. Station Setup
2. Edit Station Location
3. Set DXCC to Canada and Province to Québec
4. Log a QSO and upload to LoTW

**Expected Behavior**
Upload QSOs to LoTW

**Actual Behavior**
LoTW Error
`2024-08-03 07:07:01 LOTW_QSO: Error in record 3: Unable to verify tCONTACT signature: QSOVerify failed - Verification failed`

TQSL
[image]

QSO Uploaded from TQSL
[image]

HB9HIL, 2024-08-03:
@phl0 Could this be related to this one?

https://github.com/wavelog/wavelog/commit/4a847de5a62afecd661611b1b8f566f82a720baf

phl0, 2024-08-03:
I will have a look. Can you please provide a screenshot of the station location config in Wavelog?

anthonydiiorio, 2024-08-03:
@HB9HIL 
> Could this be related to this one?
> 
> [4a847de](https://github.com/wavelog/wavelog/commit/4a847de5a62afecd661611b1b8f566f82a720baf)

I tried commenting this out and LoTW doesn't like QC
`LOTW_QSO: Error in record 2: CA_PROVINCE: Invalid value in field (QC)`

I checked the TQSL source and Quebec is mapped to PQ as well so this should be ok. https://sourceforge.net/p/trustedqsl/tqsl/ci/master/tree/src/config.xml#l724

@phl0 
> I will have a look. Can you please provide a screenshot of the station location config in Wavelog?

[image]

anthonydiiorio, 2024-08-03:
There's definitely something up with `lotw_ca_province_map`, I tried Newfoundland and Labrador which is also covered by this function and it also breaks the upload. Then switched to a different province and it uploads normally.

HB9HIL, 2024-08-07:
Solved in the current release 1.8.1

Please test and close this issue if everything works as expected :)

anthonydiiorio, 2024-08-08:
Running smooth! Thanks 😄

PR #692 (2024-08-04): Fix LoTW signing for Quebec and Newfoundland and Labrador
Resolves #689

Adds lotw_ca_province_map to the sign_string to avoid a missmatch when signing QSOs from Quebec and Newfoundland and Labrador

[image]
[image]
closed 2024-08-08 by anthonydiiorio