wavelog/wavelog #691: Main menu's dropdown items does not collapse on small screen

filipmelik, 2024-08-03:
**Describe the bug**
When on mobile device (or device with small screen), when dropdown submenu is opened, but no item selected, and then you click some other dropdown submenu, the previous dropdown does not close. 

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- Wavelog:          1.8

**Steps to Reproduce**
1. Use small screen or developer tools
2. Click on hamburger main menu icon
3. Click QSO item
4. Click Logbook (or any other dropdown)
5. See the error

**Expected Behavior**
The previous dropdown should close. Also, all the dropdowns should close when hamburger icon is clicked to collapse the whole main menu

**Video**
[Video here](https://youtube.com/shorts/ZSej-YeDJco?feature=share)

**Desktop Information:**
- OS:               macOs, iOS, probably others as well 
- Browser:       Chrome, Safari, probably others as well

michalsiman, 2024-08-05:
Have same problem. My usage of WAVELOG is on the phone - first look, demo of WAVELOG - and i can see dropdown menu design problem. For me at this moment non-usable - wait for solve this issue.

AndreasK79, 2024-08-25:
I did look into this problem, but I could never figure out a solution. Many topics exist regarding Bootstrap and how to solve it, but for now, I was not able to get any of them to work. 

So as @HB9HIL tagged this for "help wanted", this is probably not something that will be fixed soon.

filipmelik, 2024-08-25:
Well, I also looked into this problem for a while. I have invested about hour or two, but the problem is that I am backend guy and frontend is a nightmare for me. Otherwise I would fix it myself right away. 

I have even found a working example on the net here, but was not able to get it integrated into the wavelog's source, so I gave up. Maybe you find it useful to check it out and look into it again with working example at hand possibly? :) 

Here is the [working example](https://bootstrap-menu.com/detail-multilevel.html), there is also a Demo button that will take you to the working example.

HB9HIL, 2024-08-25:
> Here is the [working example](https://bootstrap-menu.com/detail-multilevel.html), there is also a Demo button that will take you to the working example.

@AndreasK79 I think the only difference between Wavelog and the Demo Menu is that the "main" items do not open on hover, the submenu items just behave the same. Maybe this could be a solution (imho absolutely okay of the main menu does not open on hover if we fix mobile view with this)

AndreasK79, 2024-08-26:
We have found the reason for this behavior. Problem is that the code is there so that submenus works on mobile.
Now that we know why, we can hopefully tweak this.

MatykoBr, 2025-02-16:
Fixed in #1657. This can be closed!

closed 2025-02-16 by AndreasK79