wavelog/wavelog #695: eQSL Cron Job not working if password contains <

boomesito, 2024-08-05:
If eQSL password contains a "<" cron job for up- and downloading from eQSL does throw the error invalid user or password 

**Version Information**
- official Wavelog Docker image
- Wavelog: 1.8 

**Steps to Reproduce**
1. Create an eQSL account with password containing "<"
2. setup cron manager for up- and download from/to eQSL
3. enable logging in config.php
4. See error invalid user or password in logfile

**Additional Context**
- if you run the upload manually from third party services menu it works
- for debugging I logged the password and figured out, that < was converted to &lt;

HB9HIL, 2024-08-05:
@boomesito 

Do you know how to test a PR branch? :)

HB9HIL, 2024-08-06:
@boomesito 

I tested the password `test<test` in current `dev` branch at it sucessfully uploads the QSO to eQSL.cc. 

There is may another issues? 

Please change your password on eQSL and in Wavelog and provide me the password here.

boomesito, 2024-08-06:
@HB9HIL the password not working was `\Y+ZbZDF<*kJ\%`
By changing `<` to `$` it worked in currend 1.8 docker image.

HB9HIL, 2024-08-06:
Hello @boomesito 

Thx for providing the password (I hope you changed it)

I tested `\Y+ZbZDF<*kJ\%` with my own eQSL Account and the QSO's just uploaded fine. On a standard installation, also on the current docker package. 

I assume that you may had a typo in your password, which is now fixed. But I can't reproduce the issue with `\Y+ZbZDF<*kJ\%`

In my eyes this one is fixed...

boomesito, 2024-08-06:
Hello @HB9HIL 

of course I changed my password and the new one is working. I dont't believe it was a typo as it worked when in started sync from the menu. It only didn't work when update was done with the cron job.

HB9HIL, 2024-08-06:
Let me check again

HB9HIL, 2024-08-06:
Finally found the issue. Thanks for the hint again that it did no work via Cron but worked via GUI

Fix in #696

HB9HIL, 2024-08-06:
@boomesito 

issue is fixed. Will be in the next master release.

If you run your docker installation on the dev branch you can repull the image in a few minutes to test the change.

HB9HIL, 2024-08-07:
Solved in the current release 1.8.1

Please test and close this issue if everything works as expected :)

boomesito, 2024-08-07:
Works with current release

closed 2024-08-07 by boomesito