wavelog/wavelog #697: Hide Unused Columns from Logbook

stevepanaghi, 2024-08-05:
**By submitting feature requests, you have the opportunity to contribute your ideas and inspiration to the Wavelog Dev-Team. Please be aware that not every feature request will be addressed or answered.**

**What Feature is Missing?**
A way to hide unused/unwanted columns from the logbook. A second part is to then hide the data from the logbook dashboard if the column that is referenced is hidden.

**Why is This Feature Important?**
As an example, I use LotW and QRZ in place of sending a paper card. As such, I'd like to have an option to hide this column from the logbook as I don't need it, and seeing the red arrows makes me think the contact has not been confirmed. Another example is the QSL via column and QSL Msg columns. As part of this, removing these options from the overall tracking on the main logbook dashboard would allow for a cleaner dashboard to only show what each operator would like to see/uses.

**Is This Feature Personal or Beneficial to Others as Well?**
I believe this feature will benefit all users as not everyone uses all the columns, and by allowing user selection on what to hide, Wavelog become more customizable to suit each operators needs. It would also allow the logbook to be easier to read on mobile devices.

int2001, 2024-12-02:
Hi @stevepanaghi - is this still an issue?
You can disable the QSL-Methods for Overview and search at Usersettings. This simply disables the icons in the UI, and preselects the chosen Methods at the most of the analytics views.
Disabling doesn't mean that the sync stops (if sync was set-up)

[image]

stevepanaghi, 2024-12-02:
Looks like that solved my issue/request!