wavelog/wavelog #698: Logging a POTA 4fer is not possible due to the maximum length of the Database field.

clkamp, 2024-08-05:
**Describe the bug**
Logging a POTA 4fer is not possible due to the maximum length of the Database field.
A POTA 4fer with separating comma has a length of 32 chars, which is longer than the length of the COL_POTA_REF field

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               Ubuntu 20.04
- PHP:              8.1.21
- MySQL/MariaDB:    10.3.39-MariaDB
- Wavelog:          1.8

**Steps to Reproduce**
1. Go to 'Post QSO'
2. Add a QSO with 4 POTA References
3. Try to save
4. Error in GUI is 'Timeout while adding QSO. NOT added'

**Expected Behavior**
The QSO should be saved.

HB9HIL, 2024-08-05:
Hello @clkamp 
This get's fixed in the next release :)

HB9HIL, 2024-08-07:
@clkamp 
This is solved in the current release 1.8.1

clkamp, 2024-08-07:
Thanks for the fast fix, everything works as intended now 👍

PR #699 (2024-08-07): Migration 212
This migration already contains the 1.8.1 release. 

In addition to that it fixes issue 698 by resizing the POTA fields to allow 1,2,5,10-fer pota refs :) 

`fixes #698 `