wavelog/wavelog #719: SimpleFLE swaps RST Sent and received.

DC1TC, 2024-08-08:
**Describe the bug**

SimpleFLE swaps RST Sent and Received when saving it to the database.

**Steps to Reproduce**
1. Go to SimpleFLE
2. Enter a line like `20m ssb 0941 N0CALL 51 59
3. Look in the table on the right which shows it as 51 sent and 59 received.
4. Log the QSOs
5. Look at the QSOs in the Logbook. It shows it as 59 sent and 51 received.

**Expected Behavior**
RST Sent and Received are not swapped.

**Additional Context**
https://github.com/wavelog/wavelog/blob/646ccf2783042ba6390cea1ecc5ab4d3996b7b3c/assets/js/sections/simplefle.js#L350-L351
vs 
https://github.com/wavelog/wavelog/blob/646ccf2783042ba6390cea1ecc5ab4d3996b7b3c/assets/js/sections/simplefle.js#L918-L919
It may be better to use json objects rather than depending on the order in the array?

DC1TC, 2024-08-08:
Potentially caused by https://github.com/wavelog/wavelog/pull/322

DC1TC, 2024-08-08:
This bug was original found by DL2MBN, I just filed it here.

HB9HIL, 2024-08-08:
@DC1TC please test

HB9HIL, 2024-08-08:
DL2MBN confirms this bug is fixed in `dev`

closed 2024-08-08 by HB9HIL