wavelog/wavelog #726: "Gridsquare Activators" search issue with CALL/VO1 - multiple grid locators in "gridsquare"

DK9JC, 2024-08-10:
When I looked in "Gridsquare Activators" I could not found those entries from a rover from CALL/VO1.

I tracked down the issue to all those QSO were made from a gridline. Both grids logged into "Gridsquare". With most of wavelog now supporting it, it causes a "ignore" in the search results.

- PHP:              8.1.29
- MySQL/MariaDB:  10.5.25-MariaDB
- Wavelog:          1,8.2 (was before also 1.7.2 )

[image]
[image]

AndreasK79, 2024-08-12:
@DK9JC I found some problems in the code, and can reproduce this problem. Thanks for reporting! Not sure when a fix will be in place, but it's being looked into.

DK9JC, 2024-08-12:
thanks for looking. 
good it can be reproduced.

AndreasK79, 2024-08-13:
@DK9JC if you pull latest dev, the issue should now be fixed!

DK9JC, 2024-08-19:
> @DK9JC if you pull latest dev, the issue should now be fixed!

[image]

fixed, thank you!!

PR #743 (2024-08-13): This fixes activated gridsquares where only a border or corner was worked
Fixed #726.

The problem is if you work a station only in grid border or corner, it will not get listed.

I am not 100% sure that this will work with older MySQL versions.
closed 2024-08-19 by HB9HIL