wavelog/wavelog #727: Map export is not working correctly

derFogel, 2024-08-10:
If you press this button in the map of the advanced logbook:
[image]

Then the resulting image looks like this:
[image]

But I would expect something like this:
[image]

Version: 1.8

DK9JC, 2024-08-10:
I can confirm w 1.8.2

to be more specific:

the downloaded file Wavelog,png has the problem. 

the view in wavelog itself looks correct.

AndreasK79, 2024-08-10:
I can't reproduce this. Which browser are you guys using?

derFogel, 2024-08-10:
I am using firefox 127.0.2

AndreasK79, 2024-08-10:
See here: https://github.com/wavelog/wavelog/issues/432 Seems Firefox might have some trouble.

derFogel, 2024-08-10:
ok, so close as duplicate?

derFogel, 2024-08-10:
Turns out there are bug reports on leaflet-easyprint going back to 2017 with the same problem.

closed 2024-08-10 by derFogel

DK9JC, 2024-08-10:
> I can't reproduce this. Which browser are you guys using?

FF 129.0

AndreasK79, 2024-08-10:
> Turns out there are bug reports on leaflet-easyprint going back to 2017 with the same problem.

So they are aware, but no fix exists I guess. I don't remember if I checked last time it was reported, I just remembered that someone had similar problems before.