wavelog/wavelog #754: Change the Global-Language after install

NoobKeksTV, 2024-08-14:
Hi,

i set up a wavelog-instance on my server in the German language and now want it to be "public".
To be that way, i need it to be in english tho. 
I did that by changing my local user-language, yes. but i also want to share the url to my logbook and no matter which device i tried it always came back in german.

HB9HIL, 2024-08-14:
hello @NoobKeksTV 

There no "Global Language" Setting. The language of Wavelog is since version 1.7 managed by the gettext() function. 

This means:

**If a user is logged in** the language is defined by the language set in the user profile. 

**If a user is not logged in**, for example on the public visitor page or the login page, the language is set by the default language which is set in the browser. So if the default language of your browser is set to german, you see a german page. If it's per default portuguese or chinese, you see the portuguese or chinese text. 

You can test it out. Visit [https://demo.wavelog.org/](https://demo.wavelog.org/). This is our public demo. If your browser is set to german, you'll see the demo text in german. Other users, will see another language on the same page. 

Wanna test another language? No problem. Change your browsers default language to e.g. portuguese, delete all cookies and visit [https://demo.wavelog.org/](https://demo.wavelog.org/) again. You will see the portuguese text. 

Does this makes sense to you?

Vy 73 de HB9HIL

NoobKeksTV, 2024-08-14:
yes, it does. Thank you!

closed 2024-08-14 by NoobKeksTV

HB9HIL, 2024-08-14:
> yes, it does. Thank you!

Bonus: The technical part:

The language is controller by the cookie `language`

[image]

If no cookie is set on the current page (and you aren't logged in) -> `gettext` get's the default browser language and set's the cookie. -> if the cookie is available it takes the cookie's value as language. Fallback is always `english`