wavelog/wavelog #769: Update Country Files functions modify QSO data of other users

phl0, 2024-08-17:
From reading the code while coding https://github.com/wavelog/wavelog/pull/767 it seems that not only the distance function but the others ass well also modify QSO data of other users. This is potentially not intended. Needs some investigation and probably a fix.

phl0, 2024-08-17:
Bug/problem is confirmed. A QSO logged by another user is updated on calling these functions. Needs fixes.

HB9HIL, 2024-08-17:
I'm not sure if this is a bug. In fact it's an admin function and the description and usage intends to affect all QSO within Wavelog. tbh: I don't see an issue here but I'm may wrong. 

It's not that bad to have some "sanitize" functions which clean the database from missing or wrong data.

phl0, 2024-08-17:
Basic question is: Do we want to modify QSO data for any user here?

int2001, 2024-08-18:
> Basic question is: Do we want to modify QSO data for any user here?

if admin: Yes
if available for user: only Users QSOs