wavelog/wavelog #780: CronManager doesn't execute some jobs

winnieXY, 2024-08-20:
I've used the latest docker image according to [the wiki](https://github.com/wavelog/wavelog/wiki/Installation-via-Docker).

The jobs do run manually, e.g. i can upload/download to lotw, eqsl, qrz, clublog, etc. using the menu (or in case of clublog using a url). I do not find errors within the log file which could indicate a issue. Executing the command in /etc/cron/wavelog manually also doesn't indicate a error. 

```
root@0dfe98d0fcc2:/var/www/html# curl --silent http://localhost/index.php/cron/run
CRON: clublog_download -> is due: false -> Next Run: 2024-08-21 00:07:00
CRON: clublog_upload -> is due: false -> Next Run: 2024-08-20 12:03:00
CRON: eqsl_sync -> is due: false -> Next Run: 2024-08-20 12:09:00
CRON: hrdlog_upload is disabled. skipped..
CRON: lotw_lotw_upload -> is due: false -> Next Run: 2024-08-20 08:00:00
CRON: qrz_download -> is due: false -> Next Run: 2024-08-20 12:18:00
CRON: qrz_upload -> is due: false -> Next Run: 2024-08-20 12:06:00
CRON: update_dxcc -> is due: false -> Next Run: 2024-09-01 00:20:00
CRON: update_lotw_users -> is due: false -> Next Run: 2024-08-26 01:10:00
CRON: update_update_clublog_scp -> is due: false -> Next Run: 2024-08-25 00:00:00
CRON: update_update_dok -> is due: false -> Next Run: 2024-09-01 00:00:00
CRON: update_update_pota -> is due: false -> Next Run: 2024-09-01 00:15:00
CRON: update_update_sota -> is due: false -> Next Run: 2024-09-01 00:05:00
CRON: update_update_wwff -> is due: false -> Next Run: 2024-09-01 00:10:00
```

**Version Information**
- OS (Host):             Debian 12
- PHP:              not relevant due to docker
- MySQL/MariaDB:   mariadb:11.3 image
- Wavelog:     ghcr.io/wavelog/wavelog:latest (Version 1.8.2)     

**Steps to Reproduce**
1. Fill in all credentials for qrz, eqsl, lotw, ...
2. Enable all cronjobs
3. Wait 2 days
4. See attached screenshot.

[image]

As you note, some cronjobs are executed, some are not. The instance is running for two weeks already with all cronjobs enabled. (And the manual upload just works)...

**Expected Behavior**
The lotw, qrz, eqsl, ... cronjobs are executed reliable.

**Screenshots**
See above

**Desktop Information:**
- OS:               Debian Unstable
- Browser:         Firefox
- Version:          128

**Additional Context**
Please contact me in case of any questions. I'm happy to help - wavelog is awesome :-)

HB9HIL, 2024-08-20:
Hello @winnieXY 

This is really strange. Did not see this issue before. 

Did you do any modifications to the internal (real) cronjob? (edited `crontab`)

winnieXY, 2024-08-20:
Nope. 

I did not modify anything. To get it working I added right after the bugreport a old style crontab file and disabled every job within the cron manager.

Can I somehow help and debug somehow?

HB9HIL, 2024-08-20:
Can you figure out which jobs run normally and which do not?

winnieXY, 2024-08-21:
Okay - it seems as also the old-style cronjobs are not running. 
For testing purposes i did some qsos yesterday and did not upload them to lotw (expecting the cronjob to do that). 

I modified the docker compose file like this:

```
  wavelog-db:
    image: mariadb:11.3
    container_name: wavelog-db
    environment:
      MARIADB_RANDOM_ROOT_PASSWORD: Aingohnuw7ca
      MARIADB_DATABASE: wavelog
      MARIADB_USER: wavelog
      MARIADB_PASSWORD: yeeMae4phuw9 # <- Insert a strong password here
    volumes:
      - /home/docker/wavelog/data:/var/lib/mysql
    restart: always
    networks:
      wavelogbackend:
        ipv4_address: 172.174.1.20

  wavelog-main:
    container_name: wavelog-main
    image: ghcr.io/wavelog/wavelog:latest
    depends_on:
      - wavelog-db
    links:
      - web
    environment:
      - TZ=Europe/Berlin
    volumes:
      - /home/docker/wavelog/config:/var/www/html/application/config/docker
      - /home/docker/wavelog/uploads:/var/www/html/uploads
      - /home/docker/wavelog/userdata:/var/www/html/userdata
      - /home/docker/wavelog/cron/wavelogNonInternal:/etc/cron.d/wavelogNonInternal:ro
    restart: always
    networks:
      wavelogbackend:
        ipv4_address: 172.174.1.22
      webproxy:
        ipv4_address: 172.170.1.22 
        ipv6_address: 2607:5300:61:a54:2::22
    deploy:
      resources:
        limits:
          cpus: '1'
          memory: 2G

```

Please note that there is another container (called web) which is used for all instances as proxy to get https/ssl. This should have no influence on the cronjobs running inside the container.

All jobs are disabled:
[image]

Content of the wavelogNonInternal file:
```
# Upload QSOs to Club Log (ignore cron job if this integration is not required)
3 */6 * * * curl --silent http://localhost/index.php/clublog/upload &>/dev/null

7 00 * * *  curl --silent http://localhost/index.php/clublog/download &>/dev/null

# Upload QSOs to LoTW if certs have been provided every hour. (Does Up- AND Download)
0 */1 * * * curl --silent http://localhost/index.php/lotw/lotw_upload &>/dev/null

# Upload QSOs to QRZ Logbook (ignore cron job if this integration is not required)
6 */6 * * * curl --silent http://localhost/index.php/qrz/upload &>/dev/null

# Download QSOs from QRZ Logbook (ignore cron job if this integration is not required)
18 */6 * * * curl --silent http://localhost/index.php/qrz/download &>/dev/null

# Upload QSOs to HRD Logbook (ignore cron job if this integration is not required)
#12 */6 * * * curl --silent http://localhost/index.php/hrdlog/upload &>/dev/null

# Upload/download QSOs to/from Eqsl (ignore cron job if this integration is not required)
9 */6 * * * curl --silent http://localhost/index.php/eqsl/sync &>/dev/null

# Update LOTW Users Activity
# Update only once a week as data is only provided weekly at ~ 1000z Sundays
10 1 * * 1 curl --silent http://localhost/index.php/update/lotw_users &>/dev/null

# Update Clublog SCP Database File
@weekly curl --silent http://localhost/index.php/update/update_clublog_scp &>/dev/null

# Update DOK File for autocomplete
@monthly curl --silent http://localhost/index.php/update/update_dok &>/dev/null

# Update SOTA File for autocomplete
@monthly curl --silent http://localhost/index.php/update/update_sota &>/dev/null

# Update WWFF File for autocomplete
@monthly curl --silent http://localhost/index.php/update/update_wwff &>/dev/null

# Update POTA File for autocomplete
@monthly curl --silent http://localhost/index.php/update/update_pota &>/dev/null

# Update DXCC entities every two months
20 0 1 */2 * curl --silent http://localhost/index.php/update/dxcc &>/dev/null
```

Please note the timestamps in the screenshot. 
Now I log into the docker container like this and execute the (e.g.) clublog_upload manually.

``` 
sudo docker exec -it wavelog-main bash                                                                                                                  
root@522a35066666:/var/www/html# curl --silent http://localhost/index.php/clublog/upload
Clublog upload for DJ1PW
root@522a35066666:/var/www/html#
```
Now the timestamp within the cron-list is modified:

[image]

So indeed - this seems to work, however the complete cron system seems to be disfunctional in my case within the container.

Any clues?

winnieXY, 2024-08-21:
I've create a simple test:

```
root@c2b208e09718:/tmp# cat /etc/cron.d/wavelogNonInternal 
#Create a dummy test file to check for cron execution.
* * * * * echo "$(date)" >> /tmp/test
```

I've added this simple stanza in front of my custom cron file and recreated the container. 
/tmp is filling up with ci_waveflog* files, however there is no test file there:

```
root@c2b208e09718:/tmp# ls /tmp/
ci_wavelog015844f0c39f8e3d800bb98ee1592257aa2b4cd2  ci_wavelog4ad9207be2a7d8b869817f8d9a1e2bf19400fe4a	ci_waveloga389cdda710a89ef73d6584b1fbba4bac0f267aa
ci_wavelog084435d8c1544ae7655d27babda016a57059e7ef  ci_wavelog53daf970c59a170dde0be173d5b262445dc7600a	ci_wavelogac7b3114d6044c571e4a42e2eb46cafb8f1b4f7c
ci_wavelog0bcdf9a2238687261e9782934bc710f1e119a4e3  ci_wavelog55d5ba9963f3b192a28fd3496632284094cdf886	ci_wavelogb748d9c567de2c5bf3b41422f73bdadb96df1ee9
ci_wavelog0d3bd42136076c27b37e141558c34702f7e4c8ff  ci_wavelog5d578b99458b9338d62f11340e13819faa2a7952	ci_wavelogb89826e5af7afd076f260e4fb6ae3e581f601324
ci_wavelog1244cd8845175bc40eb8c488dcb335b7a09b5343  ci_wavelog67e0321cd6f5d454b60f5b45a7de7e248daf18fe	ci_wavelogb8d3894236242c39104ea915f9de5135d9dc8793
ci_wavelog1976589ed946f675b8562f1695144232f973fce6  ci_wavelog684b996255847de96cdd35e457fbf536482a27a1	ci_wavelogbd8c4a8edc47fc5d181c310696f94a64972cc1bb
ci_wavelog207f908352e50e31eea3bc1478b07d187378bca6  ci_wavelog71d355097c651cb62ba8bb915939a7e3ecaf6641	ci_wavelogbec05aca57219fd54584dd60fd9c1d2762d4bbcf
ci_wavelog25b2798ba08c9b11fd05333c0706ca159b07b653  ci_wavelog765d3d2c475746fde288f16e7ef9cc17ac94b37e	ci_wavelogc00f075c3298a36d3ba7ec0faaac5512f132a31c
ci_wavelog2d8040f01f75fc82b7f3faece4eda8f3d112fadb  ci_wavelog794056937c84ea91bc94598c461e2c75d21d55a5	ci_wavelogd742eb514069c1e8d0ce49cbd50b324074579eb6
ci_wavelog353e6997fd3d6eae6fc83a964465f11f41d92b25  ci_wavelog7b35168dc54e863a510f0a7b46c5b43c12470481	ci_wavelogdf524b29ef09696b1e6eb93bdb67e3c3f1738dfd
ci_wavelog36837f6ce0f0e371959b72cbba204bb7ea86646a  ci_wavelog8d279bceec4d55c9a764fd81c6dc065af9f919f9	ci_wavelogf74e94efa68677f74e9c3a9424b50d994261f23b
ci_wavelog3e2dfc416d66495869f08477573fa65ad79f888f  ci_wavelog96320f91f6d604a1a027373397b2a8205e8caccc	ci_wavelogffbe7cd1f779163973ec217801ae9f43c979dd02
ci_wavelog405df3a45b71a93a7c32d352f8022e30af265a44  ci_wavelog9b349bf5d80de9b4a201c8fc3fe2712fddff3fbc
root@c2b208e09718:/tmp#
```

It looks like the complete cron is not running at all and my first impression (some cronjobs are running) was false.

HB9HIL, 2024-08-21:
You whole setup is far away from a standard docker compose setup as we support it. 

The Cronmanager works like this

There is ONE real cronjob which calls `http://localhost/index.php/cron/run` once per minute `* * * * *`

Each time the cron is running it checks if one of the enabled cronjobs in the cronmanager is due (based on the calculated next run -> has to match with current time). That's it . In fact the cronmanager does not handle "real" cronjobs due the nature of PHP.

HB9HIL, 2024-08-21:
I tested the cronmanager in various combinations on `master` and `dev` branch, in and outside of docker. I can't spot any issues to this point. The fact that your whole setup is really far away from our provided standard setup makes it quite obvious that there is something wrong in your setup. So for me this is no bug in Wavelog at the moment.

winnieXY, 2024-08-21:
> You whole setup is far away from a standard docker compose setup as we support it.
>
> The Cronmanager works like this
>
> There is ONE real cronjob which calls `http://localhost/index.php/cron/run` once [...]

Yes of course. 
In fact the only difference of my initial setup was/is the proxy before wavelog (which needs wavelog to be in a separate subnet). 
All the other stuff written above is just added to debug the issue and I would be more than glad to get rid of it. :)

winnieXY, 2024-08-21:
I've replaced now the installed crontab in /var/spool/cron/crontabs/root with basically the file shown above. If this works than I'm sure that cron itself is not a issue but something else.

winnieXY, 2024-08-21:
> Can you figure out which jobs run normally and which do not?

Do you have any hints how I can debug the container in a other way without changing so much? :)

HB9HIL, 2024-08-21:
@winnieXY 

I see several Issues in this setup

1. You need the environment variable `CI_ENV: docker`. Otherwise the config does not live in `application/config/docker` and will be overwritten on repull
2. You don't need `TZ=Europe/Berlin`. Whole Wavelog is working in UTC and we can't say if this can cause issues if not

To debug it more easy you could create another volume for the application logs
`- /home/docker/wavelog/logs:/var/www/html/application/logs`

HB9HIL, 2024-08-21:
And as already said: Only **one** cronjob is necessary
```
* * * * * curl --silent http://localhost/index.php/cron/run &>/dev/null
```

winnieXY, 2024-08-21:
> @winnieXY
> 
> I see several Issues in this setup
> 
>     1. You need the environment variable `CI_ENV: docker`. Otherwise the config does not live in `application/config/docker` and will be overwritten on repull

Added that

> 
>     2. You don't need `TZ=Europe/Berlin`. Whole Wavelog is working in UTC and we can't say if this can cause issues if not

Deleted that. 
 
> 
> To debug it more easy you could create another volume for the application logs `- /home/docker/wavelog/logs:/var/www/html/application/logs` and enable logging in `application/config/docker/config.php` -> `$config['log_threshold'] = 1` (error level is enough)

left untouched as it is =2 currently.

> 
> Important is: In `application/config/docker` are only **2 files** config.php database.php all other config files live in `application/config` and should not be touched

this is the case.

winnieXY, 2024-08-21:
I'll report back this evening or tomorrow.

After reviewing the changes from beginning till here the main change is the missing `CI_ENV: docker ` environmental variable.

winnieXY, 2024-08-21:
Okay..
Now i got at least a error within the CronManager:
[image]

This was supposed to run at 12:03:00. The mounted logs folder is however empty. 
After another minute the Status switched back to "healty" btw.

HB9HIL, 2024-08-21:
> 12:00 is written behind 12:02, 12:03 is behing 12:05 and 12:09 is behind 12:11.

@int2001 any idea? :D

winnieXY, 2024-08-21:
Just some code, without testing (yet) - maybe something like this?
```
#!/bin/bash
# Check is Lock File exists, if not create it and set trap on exit
if { set -C; 2>/dev/null >/var/lock/wavelogcron.lock; }; then
        trap "rm -f /var/lock/wavelogcron.lock" EXIT
else
        echo "Lock file exists… exiting"
        exit
fi
curl --silent http://localhost/index.php/cron/run 
```
And call than this small shell script instead of curl within the cron

int2001, 2024-08-21:
> for me it might look like a race condition ;-) I'm currently thinking about a way to prevent the cronjob from running when another curl instance is still running.

yep. race-cond.
do you have huge amounts to sync?

in rare cases it may be, that the "old" cron is still running/working, while a new one comes around.
And after the first one is finished some random one from the queue will be processed.

you can try working without the cronmanager (disable the 1min. job and enable the single-job).
or: you wait until everything is processed and keep an eye on it.

i wasn't able to repro it.

HB9HIL, 2024-08-21:
> Just some code, without testing (yet) - maybe something like this?
>
> ```
> #!/bin/bash
> # Check is Lock File exists, if not create it and set trap on exit
> if { set -C; 2>/dev/null [...]

This wont work because if the nature how the cronmanager works.

winnieXY, 2024-08-21:
> yep. race-cond.
> do you have huge amounts to sync?
>
> in rare cases it may be, that the "old" cron is still running/working, while a new one comes around.
> And after the first one is finished some [...]

What do you think about my proposed snipplet to prevent the race condition?

As this is within the docket container all modifications to cron are somehow difficult compared to a normal install.

HB9HIL, 2024-08-21:
The current cronmanager calls the "internal" stuff with curl aswell (unfortunately) -> causes curl in curl

This is no issue as long everything stays close together (local network, which is "localhost")

There is a progress tracker #339. I move executive stuff from the controllers to the models. If this is finished we can call the internal stuff directly in PHP without the need of curl

winnieXY, 2024-08-21:
> Try the following:
> 
> call the cron/run from outside of the docker container. from an external machine as minute interval cronjob
> 
> `* * * * * curl --silent https://public-url/cron/run &>/dev/null`
> 
> if this solves the issue, it's probably your cron setup

I'll test it till this evening with the following setup:

```
wavelog-main:
    container_name: wavelog-main
    image: ghcr.io/wavelog/wavelog:latest
    depends_on:
      - wavelog-db
    links:
      - web
    environment:
      #      - TZ=Europe/Berlin
      - CI_ENV=docker
    volumes:
      - /home/docker/wavelog/config:/var/www/html/application/config/docker
      - /home/docker/wavelog/uploads:/var/www/html/uploads
      - /home/docker/wavelog/userdata:/var/www/html/userdata
      - /home/docker/wavelog/logs:/var/www/html/application/logs
      - /home/docker/wavelog/cron/wavelogNonInternal:/var/spool/cron/crontabs/root:ro
    restart: always
    networks:
      wavelogbackend:
        ipv4_address: 172.174.1.22
      webproxy:
        ipv4_address: 172.170.1.22 
        ipv6_address: 2607:5300:61:a54:2::22
    deploy:
      resources:
        limits:
          cpus: '1'
          memory: 2G
```
This ro-mounted crontab file is just empty, thus disabling the internal crontab/ curl call.

winnieXY, 2024-08-21:
Externally called:

[image]

Please note that I increased the interval for debugging from */6 to */1 (the /1 is just a reminder for me to change it back afterwards).

winnieXY, 2024-08-21:
I disabled also the remote call to curl and executed the curl instead in a bash environment within the docker container:

```
CRON: clublog_download -> is due: false -> Next Run: 2024-08-22 00:07:00
CRON: clublog_upload -> is due: true
CRON: clublog_upload -> RUNNING...
ERROR: Something went wrong with clublog_upload
CRON: eqsl_sync -> is due: false -> Next Run: 2024-08-21 18:09:00
CRON: hrdlog_upload is disabled. skipped..
CRON: lotw_lotw_upload -> is due: false -> Next Run: 2024-08-21 15:00:00
CRON: qrz_download -> is due: false -> Next Run: 2024-08-21 18:18:00
CRON: qrz_upload -> is due: false -> Next Run: 2024-08-21 18:06:00
CRON: update_dxcc -> is due: false -> Next Run: 2024-09-01 00:20:00
CRON: update_lotw_users -> is due: false -> Next Run: 2024-08-26 01:10:00
CRON: update_update_clublog_scp -> is due: false -> Next Run: 2024-08-25 00:00:00
CRON: update_update_dok -> is due: false -> Next Run: 2024-09-01 00:00:00
CRON: update_update_pota -> is due: false -> Next Run: 2024-09-01 00:15:00
CRON: update_update_sota -> is due: false -> Next Run: 2024-09-01 00:05:00
CRON: update_update_wwff -> is due: false -> Next Run: 2024-09-01 00:10:00

```
It took ages till it printed something and than I got that error there.

HB9HIL, 2024-08-21:
Just a fun question: Do you use a valid SSL certificate or a self-signed?

HB9HIL, 2024-08-21:
@winnieXY 

I tested your suspicion about race-conditions.

[image]

I was able to simulate a really long running PHP function (`sleep(600)`) which resulted in the access log you saw aswell. 

In fact it does not affect the cronmanager or the cron/run function as curl opens a new session everytime it gets called in PHP (CLI has no cookies). 

[image]

The result is, that the functions get called when the should get called. No race conditions here, they run in parallel.

winnieXY, 2024-08-21:
> Just a fun question: Do you use a valid SSL certificate or a self-signed?
> 
> To specify
> 
> The URL `base_url` in your config.php. Does this URL have a valid SSL certificate or is it just a self-signed?

Hey,

it's a let's encrypt certificate which contains a lot of subdomains (thus the webproxy in between). the base url matches the url within the certificate. 

_However_ as the local curl accesses  the instance via localhost and unencrypted - this might be a problem? The instance itself doesn't know anything about https, except the base_url.

HB9HIL, 2024-08-25:
Should be fixed now with #795  :)

closed 2024-08-25 by HB9HIL