wavelog/wavelog #788: Automatically Create a new Location during ADIF Import

Loki1737, 2024-08-21:
**What Feature is Missing?**
An ADIF File provides all the information needed to create a station location. Nevertheless do I habe to create manually the station location, before I could import the ADIF file. 

**Why is This Feature Important?**
It would be a great help, if the Location is automatically created based on the information in the file, this reduces manual work and improves the automation. 

**Is This Feature Personal or Beneficial to Others as Well?**
I believe it is helpful for all people who are working at a couple of different locations. So especially for the POTA & SOTA Community. 

Due to the experted impact regarding the implementation work, I understand if this will have to wait some time in the backlog. 

BR,
DO5LE
Lucas

DocViper666, 2024-12-18:
Full support from my side as a SOTA activator. 
It is just impossible to add hundreds of locations (for each unique summit I have activated) manually.
So I can see an option to split (group-by like) and imported adif by certain conditions (MY_SOTA, MY_POTA, MY_GRIDSQUARE+STATION_CALLSIGN) and list the new locations to be created.

Or provide an API for creation of locations (or locations with their qsos). I would then created an importer tool to push the locations and qso from one or more adif files. I think that would be actually preferable? 

73 Joe, OE5JFE

remy56k, 2024-12-25:
I agree, in /P (or /M), it's weird to create a station location for each different grids. 

I have tried to keep only one location for my /P (because I always setup the correct current grid in WSJT and each grids are stored/sent in the QSO), but all Wavelog entries are referenced to a fixed single grid defined in the profile, not the imported QSO grids. And some ADI files can have many grids (JT8 roadtrip :smile: ).

Maybe it's a better option to rely on the QSO grid and not on the fixed one ? It will be more permissive and accurate, and let people to manage one CALLSIGN/P with many grids instead create hundred of "Station Locations", and switch in WaveLog-Gate. It could be an option in "Station Locations" : no set the grid = rely on the QSO grid.

Thanks, 
73

JayToTheAy, 2026-01-20:
It could maybe be done as a checkbox on the ADIF upload page? If a user checks it, whatever data is present for location (WWFF, POTA, IOTA, Grid, eQSL Nickname, etc.) on the **first** QSO record in the log is used to create a new location and this new location is used for that upload. Something like this?

[image]

I agree that picking which location to use in a potentially contradictory log is a bit arbitrary, but I think first makes it something that a user can easily check by just opening the file, and as long as it's clearly communicated to a user, it would do a lot to simplify the onboarding process when you're coming from a logger that doesn't care about location.