wavelog/wavelog #799: Contest-Map isnt working.

NoobKeksTV, 2024-08-24:
My Contest-Calender Page gives me a Error 500. After turning WaveLog into Developer mode, it shows me the following error:
https://img.egopvp.com/images/As8M.png

No error-logs inside of any apache logging found, only this error shown.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:              Apache 2.4.38 , Debian 10
- PHP:              8.3.6
- MySQL/MariaDB:    10.5.24-MariaDB-1:10.5.24+maria~deb10
- Wavelog:          1.8

**Steps to Reproduce**
1. Log into WaveLog.
2. Enable ALL Contest.
3. Goto Tools -> Contest Calender

**Expected Behavior**
To load the page.
It worked when i first installed WL, idk what i did to break it, i added my own contest and disabled all other. After it broke i went and enabled all of them again and it still wont work.

**Desktop Information:**
- OS:               Windows 10
- Browser:          Chrome 
- Version:          128.0.6613.84 (Offizieller Build) (64-Bit) (cohort: M128 Rollout) 

If needed, i can give access to some data / db.

NoobKeksTV, 2024-08-24:
Error from WL log (just found it lol):
ERROR - 2024-08-24 19:15:10 --> Severity: error --> Exception: Call to a member function format() on false /var/www/cdn/wavelog/application/controllers/Contestcalendar.php 148

HB9HIL, 2024-08-24:
Dupe #773

closed 2024-08-24 by HB9HIL