wavelog/wavelog #803: Report basic errors in Simple Fast Log Entry (SFLE) right away

filipmelik, 2024-08-25:
**What Feature is Missing?**
When you use the FLE functionality for logging QSO's (as I do when I am on portable) and you do not specify the basic information, like band or mode, the error is reported to you when "Save in Wavelog" button is pressed. It would be cool to know this information up-front, even before you start the logging, just to improve the UX a little. Especially, when the source code is ready for it (but there is a little bug in it, I plan to inform the upstream source author OK2CQR about this and fix it on his side too).

**Steps to Reproduce**
1. Go to 'QSO > Simple Fast Log Entry'
2. Paste following to the textarea: 
ok12asd
adas3sda
3. See that no error is displayed, even though the basic info about the band/mode/time is missing

**Expected Behavior**
Errors should be presented to user right away to know he should add the basic data

**Why is This Feature Important?**
I believe this would improve the FLE entry's UX

**Is This Feature Personal or Beneficial to Others as Well?**
I hope so :)

**Screenshot of the current state**
[image]

**Screenshot of the proposed state**
[image]

**Additional Context**
I have already fixed it on my side. If you think this is beneficial to others I will submit a PR.

HB9HIL, 2024-08-25:
Go ahead 😁

filipmelik, 2024-08-25:
Aye aye captain 😁 https://github.com/wavelog/wavelog/pull/804

PR #804 (2024-08-25): #803 - Add SFLE JS errors
Closes #803
closed 2024-08-25 by HB9HIL