wavelog/wavelog #843: Internal Server Error due to the mod_rewrite

toseppo, 2024-08-30:
```
[Fri Aug 30 07:05:47.791655 2024] [core:trace5] [pid 826:tid 826] protocol.c(713): [client 192.168.0.2:50903] Request received from client: POST /user/login HTTP/1.1
[Fri Aug 30 07:05:47.792054 2024] [http:trace4] [pid 826:tid 826] http_request.c(435): [client 192.168.0.2:50903] Headers received from client:, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792110 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Host: 192.168.0.1, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792134 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792157 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Accept-Encoding: gzip, deflate, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792177 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Accept-Language: en-GB,en;q=0.9, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792198 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Content-Type: application/x-www-form-urlencoded, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792218 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Origin: http://192.168.0.1, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792239 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.6 Safari/605.1.15, referer: http://192.168.0.1/user/logi
n
[Fri Aug 30 07:05:47.792262 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Connection: keep-alive, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792282 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Upgrade-Insecure-Requests: 1, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792301 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Referer: http://192.168.0.1/user/login, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792322 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Content-Length: 41, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.792342 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Cookie: ci_wavelog=ef008cd231440cce80305e599c817a9c535c6231; language=english; ci_cloudlog=b81bc8573ebfc3def1771a231402dbbf84aab101, referer: http://192.168.0.1/user/lo
gin
[Fri Aug 30 07:05:47.793084 2024] [authz_core:debug] [pid 826:tid 826] mod_authz_core.c(815): [client 192.168.0.2:50903] AH01626: authorization result of Require all granted: granted, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.793117 2024] [authz_core:debug] [pid 826:tid 826] mod_authz_core.c(815): [client 192.168.0.2:50903] AH01626: authorization result of <RequireAny>: granted, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.793139 2024] [core:trace3] [pid 826:tid 826] request.c(360): [client 192.168.0.2:50903] request authorized without authentication by access_checker_ex hook: /user/login, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.793239 2024] [rewrite:trace3] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] add path info postfix: /var/www/html/user -> /var/www/html/user/login, referer
: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.793303 2024] [rewrite:trace3] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] strip per-dir prefix: /var/www/html/user/login -> user/login, referer: http://
192.168.0.1/user/login
[Fri Aug 30 07:05:47.793333 2024] [rewrite:trace3] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] applying pattern '^(.*)$' to uri 'user/login', referer: http://192.168.0.1/us
er/login
[Fri Aug 30 07:05:47.793425 2024] [rewrite:trace4] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] RewriteCond: input='/user/login' pattern='^/backup/$' => not-matched, referer:
 http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.793461 2024] [rewrite:trace3] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] add path info postfix: /var/www/html/user -> /var/www/html/user/login, referer
: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.793491 2024] [rewrite:trace3] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] strip per-dir prefix: /var/www/html/user/login -> user/login, referer: http://
192.168.0.1/user/login
[Fri Aug 30 07:05:47.793554 2024] [rewrite:trace3] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] applying pattern '^(.*)$' to uri 'user/login', referer: http://192.168.0.1/us
er/login
[Fri Aug 30 07:05:47.793607 2024] [rewrite:trace4] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] RewriteCond: input='/var/www/html/user' pattern='!-f' => matched, referer: htt
p://192.168.0.1/user/login
[Fri Aug 30 07:05:47.793650 2024] [rewrite:trace4] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] RewriteCond: input='/var/www/html/user' pattern='!-d' => matched, referer: htt
p://192.168.0.1/user/login
[Fri Aug 30 07:05:47.793682 2024] [rewrite:trace5] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] escaping backreference 'user/login' to 'user/login', referer: http://192.168.1
.40/user/login
[Fri Aug 30 07:05:47.793711 2024] [rewrite:trace2] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] rewrite 'user/login' -> '/index.php?/user/login', referer: http://192.168.0.1
/user/login
[Fri Aug 30 07:05:47.793742 2024] [rewrite:trace3] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] split uri=/index.php?/user/login -> uri=/index.php, args=/user/login, referer: http://192.168.0.1/use
r/login
[Fri Aug 30 07:05:47.793771 2024] [rewrite:trace2] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] setting lastsub to rule with output /index.php?/$1, referer: http://192.168.1.
40/user/login
[Fri Aug 30 07:05:47.793803 2024] [rewrite:trace2] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] trying to replace context docroot /var/www/html with context prefix , referer:
 http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.793832 2024] [rewrite:trace1] [pid 826:tid 826] mod_rewrite.c(504): [client 192.168.0.2:50903] 192.168.0.2 - - [192.168.0.1/sid#b2f0c398][rid#b68e7058/initial] [perdir /var/www/html/] internal redirect with /index.php [INTERNAL REDIRECT], referer: http://192.168
.1.40/user/login
[Fri Aug 30 07:05:47.794400 2024] [authz_core:debug] [pid 826:tid 826] mod_authz_core.c(815): [client 192.168.0.2:50903] AH01626: authorization result of Require all granted: granted, referer: http://192.168.0.1/user/login
[Fri Aug 30 07:05:47.794446 2024] [authz_core:debug] [pid 826:tid 826] mod_authz_core.c(815): [client 192.168.0.2:50903] AH01626: authorization result of <RequireAny>: granted, referer: http://192.168.0.1/user/logi [... cut ...] www/html/] pass through /var/www/html/index.php, referer: http://192.168.0.1/use
r/login
[Fri Aug 30 07:05:48.243719 2024] [http:trace3] [pid 826:tid 826] http_filters.c(1141): [client 192.168.0.2:50903] Response sent with status 500, headers:, referer: http://192.168.0.1/user/login
```

After entering username, password and pressing login button, error page is displayed (server returns 500). Though navigating  manually to the `/` in the browser shows application page with the user being logged.

HB9HIL, 2024-08-30:
Hello

I assume you want to remove index.php from your URL?

Did you then
- set `$config['index_page'] = 'index.php';` to `$config['index_page'] = '';` in your config.php
- enabled the rewrite module of apache2 -> `a2enmod rewrite`
- Created the `.htaccess` file based on the sample? -> `cp htaccess.sample .htaccess`

HB9HIL, 2024-08-30:
In addition to that you may have to add this in your apache2 site config:

```
        <Directory /var/www/html>
                Options Indexes FollowSymLinks MultiViews
                AllowOverride All
                Require all granted
        </Directory>
```

toseppo, 2024-08-30:
> Hello
>
> I assume you want to remove index.php from your URL?
>
> Did you then
>
> * set `$config['index_page'] = 'index.php';` to `$config['index_page'] = '';` in your config.php
> * enabled the rewrite [...]

Yes, this is done like that.

HB9HIL, 2024-08-30:
> In addition to that you may have to add this in your apache2 site config:
> 
> ```
>         <Directory /var/www/html>
>                 Options Indexes FollowSymLinks MultiViews
>                 AllowOverride All
>                 Require all granted
>         </Directory>
> ```

and this aswell?

toseppo, 2024-08-30:
> > In addition to that you may have to add this in your apache2 site config:
> > ```
> >         <Directory /var/www/html>
> >                 Options Indexes FollowSymLinks MultiViews
> >                 AllowOverride All
> >                 Require all granted
> >         </Directory>
> > ```
> 
> and this aswell?

Added, still same error with 500.

toseppo, 2024-08-30:
I did use ghcr.io/wavelog/wavelog 1.8.2, 1.8.2 and custom build image - all result in mod_rewrite error.

HB9HIL, 2024-08-30:
> I did use ghcr.io/wavelog/wavelog 1.8.1, 1.8.2 and custom build image - all result in mod_rewrite error.

Wait you are using docker?

toseppo, 2024-08-30:
> > I did use ghcr.io/wavelog/wavelog 1.8.1, 1.8.2 and custom build image - all result in mod_rewrite error.
> 
> Wait you are using docker?

Yes, running via compose.

HB9HIL, 2024-08-30:
> > > I did use ghcr.io/wavelog/wavelog 1.8.1, 1.8.2 and custom build image - all result in mod_rewrite error.
> > 
> > 
> > Wait you are using docker?
> 
> Yes, running via compose.

**Well this might a important information :D** 

Make sure your `docker-compose.yml` looks EXACTLY the same like [this](https://github.com/wavelog/wavelog/wiki/Installation-via-Docker#docker-composeyaml) one.

Also make sure you have only **2** files in your mapped config volume. 

config.php
database.php

toseppo, 2024-08-30:
> **Well this might a important information :D**
>
> Make sure your `docker-compose.yml` looks EXACTLY the same like [...]

Yes it is like that except some changes to the image of mariadb.

toseppo, 2024-08-31:
Here is the solution that works for me: https://github.com/wavelog/wavelog/pull/850

HB9HIL, 2024-09-02:
Hello @toseppo 

I just checked your PR. See there for PR related details

I still think that your docker compose is somehow broken because of adjustments you made. I told you to edit the apache2 config before I knew you are using docker (may give a full system description in the future when creating an issue). In fact we have several docker users and no one has the same issue like you. So it must be something in your local setup.

To properly break this down I would need some more information. It's really strange for me that your PR is fixing your issue as in fact it has zero effect on the behaviour of the language cookie. The language cookie already has a expiretime of 1 month and even if it expires, it just gets recreated.

I just read again your debug log you posted above and found one thing which may is somehow an eyecatcher:

> [Fri Aug 30 07:05:47.792342 2024] [http:trace4] [pid 826:tid 826] http_request.c(438): [client 192.168.0.2:50903]   Cookie: ci_wavelog=ef008cd231440cce80305e599c817a9c535c6231; language=english; ci_cloudlog=b81bc8573ebfc3def1771a231402dbbf84aab101, referer: http://192.168.0.1/user/login

It seems here are two session cookies existing. One for Wavelog and One for Cloudlog. Because the changes you made in your PR do not affect any session cookies. It's _only_ the language cookie which shouldn't be relevant for login in any kind. 

You may want to try the following: Delete all cookies and clear the cache. Restart the container without any modifications and try again. 

See the PR for PR related details.

toseppo, 2024-09-02:
Thanks for the insights. During debugging I've tried to delete all the cookies and start in clean browser - problem was persistent. Left some comment on PR .

HB9HIL, 2024-09-07:
Should be fixed

closed 2024-09-11 by toseppo