wavelog/wavelog #846: Option to Hide Login from Vistor Page

desertblade, 2024-08-30:
**What Feature is Missing?**
Have the option to remove/hide the login button from visitor pages

**Why is This Feature Important?**
Having the option to remove the button hides an attack layer and in my case not show a 'bad user experience" if someone clicks it.

I plan to expose  "/visitor/*" through NPM reverse proxy using Cloudflare. For access to the rest of Cloudlog I will be either using VPN or a ZeroTrust to access remotely.

**Is This Feature Personal or Beneficial to Others as Well?**
Yes

Its been awhile since I worked in PHP, but this is something I could contribute.  I imagine a check box in the *Edit visitor link* popup.

phl0, 2024-08-30:
Feel free to contribute code for that. IMHO it just *hides* an attack layer and nothing more. Will not *prevent* abuse. 😇

HB9HIL, 2024-08-30:
>  I imagine a check box in the _Edit visitor link_ popup.

The Github button is managed in the global options -> appearance. Such a switch should probably live there aswell as this is more a global setting.

closed 2024-09-18 by github-actions[bot]