wavelog/wavelog #854: Band gets reset when editing an QSO

maciejla, 2024-08-31:
Hi, I found this error - not sure if it happens on all installs, just docker installs, or maybe just for me, but I figured it won't hurt to report it :) 

**Version Information**
- Environment: docker
- OS: Docker container (host is Alpine 3.20)
- PHP: `8.3.10`
- MySQL/MariaDB: `11.3.2-MariaDB-1:11.3.2+maria~ubu2204`
- Wavelog: `1.8.2`

**Steps to Reproduce**
1. Go to `Logbook` -> `Overview`
2. Click on any callsign in the log
3. Click on `Edit QSO`
4. Band gets reset from whatever was set previously to the first one in the list (by default, 160m)

**Expected Behavior**
Band stays the same when clicking on `Edit QSO`

**Desktop Information:**
- OS: Fedora 40
- Browser: Firefox 129.0.2

int2001, 2024-09-01:
Was not able to reproduce it.

e.g.: 

1. Default Band is 40m
2. Log QSO on 20m
3. Go to overview and click on QSO.
4. At popup click Edit QSO
5. Band is still 20m

Even after saving it and doing point 3 and 4 again, it stays at 20m

phl0, 2024-09-01:
Same here. Band does not change upon edit. Even for SAT QSOs with BAND and BAND_RX.

HB9HIL, 2024-09-01:
> * Wavelog: `1.8.2`

You may want to repull the latest image to upgrade to 1.8.3. Maybe this solves your issue.

maciejla, 2024-09-02:
Updated the image, but it did not help :( I'll re-install everything on a different host and see if the same issue occurs

HB9HIL, 2024-09-02:
Clear your browser cache, cookies and history. Make sure your browser is absolutely clean.

int2001, 2024-09-07:
i put this thing on Autoclose since there's no more activity.

closed 2024-09-16 by github-actions[bot]

maciejla, 2024-10-26:
Seems like it fixed itself after some time... weird, but it works now so that's great :)