wavelog/wavelog #864: PWA on Andoid locked to portrait mode

Beiri22, 2024-09-03:
Hi,

I have tried to install wavelog (as PWA) on anroid. But the PWA is locked to portrait mode, I cannot get it to work in landscape mode. On tablet compouters the portrait mode is quite nice.

HB9HIL, 2024-09-03:
To make it a bit clearer. This is not a "real" PWA. A real PWA would work also when using it offline. This is just a browser without any browser features. 

I tested the landscape PWA mode on several devices when I implemented it. On Android and iPhones it does break some modal dialogs and therefore I decided to fix the orientation to portrait mode. 

Wavelog is not build to use it on mobile. We would love to follow the "Mobile First"-Line in development but this is not possible with our current codebase. 

I'm sorry.

Beiri22, 2024-09-04:
I know about the nature of this installation. I prefer it, because there is no address bar. On the tablet PC the application seems quite ok in landscape mode(via an browser).

Do these model dialogs only break in pwa mode or also when using the application in the browser in landscape mode on Android?

HB9HIL, 2024-09-04:
You can try it yourself 

It should be **Line 7** in `manifest.json`. Simply remove this line and "reinstall" the PWA

closed 2024-09-13 by Beiri22