wavelog/wavelog #878: Unable to upload using APIs.

N3evin, 2024-09-05:
Latest 1.8.3 version. Have no issue on 1.8.2.

```
INFO - 2024-09-05 01:28:32 --> Model "Stations" initialized
INFO - 2024-09-05 01:28:32 --> Model "Oqrs_model" initialized
INFO - 2024-09-05 01:28:32 --> Controller Class Initialized
INFO - 2024-09-05 01:28:32 --> Model "API_Model" initialized
INFO - 2024-09-05 01:28:32 --> Model "User_Model" initialized
INFO - 2024-09-05 01:28:32 --> Model "Logbook_model" initialized
ERROR - 2024-09-05 01:28:32 --> Severity: error --> Exception: Data too long for column 'COL_CONT' at row 3 /var/www/html/system/database/drivers/mysqli/mysqli_driver.php 301
INFO - 2024-09-05 01:28:33 --> Config Class Initialized
INFO - 2024-09-05 01:28:33 --> Hooks Class Initialized
INFO - 2024-09-05 01:28:33 --> UTF-8 Support Enabled
INFO - 2024-09-05 01:28:33 --> Utf8 Class Initialized
```

int2001, 2024-09-05:
Good morning @N3evin .
pse check your QSOs. it seems like you tried to import/log a QSO with a very long (too long) String for the ADIF-Field CONTinent.

would be interesting how the original QSO looks like

N3evin, 2024-09-05:
> Good morning @N3evin . pse check your QSOs. it seems like you tried to import/log a QSO with a very long (too long) String for the ADIF-Field CONTinent.
> 
> would be interesting how the original QSO looks like

Don't see any adnormal info in the ADIF. I am using sdr-control to automate it. Which had no issue with the previous version.

closed 2024-09-05 by N3evin

int2001, 2024-09-05:
perhaps we're implementing to truncate to the field, so it's more failsafe.
tu for the hint/issue.