wavelog/wavelog #900: still lotw upload issue?

Beiri22, 2024-09-06:
Hi,

I just updated the server and saw that only 12 of my 58 logs have been added to lotw. So I just set all qsos to not uploaded and did the lotw sync (as in the release message). Wavelog showed an ok message and marked all qsos as uploaded. But in lotw i only have 18 now. What am I missing? IS there still a bug; or are my sqos invalid? What makes them invalid for lotw? Is there any way to display some kind of debug information when syncing to lotw?

phl0, 2024-09-06:
Which version are you using? We just released 1.8.5 which fixes another LoTW issue ...

Beiri22, 2024-09-06:
1 hr ago it was 1.8.4

Beiri22, 2024-09-06:
I'll have a try

Beiri22, 2024-09-06:
Yes, that did the trick. Thx & 73

closed 2024-09-06 by Beiri22