wavelog/wavelog #912: Portable-Mode or Portable-Station

Beiri22, 2024-09-09:
I'm curious how someone operating from multiple locations could efficiently set up Wavelog. Creating a new station for each location seems like it would clutter the station list quickly. Could there be a "portable mode" instead? For example, by selecting a "portable" checkbox in the QSO input dialog, your callsign could automatically be modified to include .../p, and additional fields could appear for setting your current location or locator. For those who are highly portable-active, this could be a great way to avoid overcrowding the station list while keeping everything organized. (Or is there already a way that I did not notice?)

HB9HIL, 2024-09-09:
I always create a new station location if I never operate from this location. No problem to have a lot of different locations.

Beiri22, 2024-09-09:
Thanks, @HB9HIL, for your perspective! While it's true that creating new station locations for every unique spot is a valid approach, I think a portable mode could enhance the user experience for operators who move frequently and cusually between locations. While some users may be okay with a large number of locations, others -like me- might find that overwhelming.  A portable mode checkbox could streamline this, reducing the need to set up multiple entries, especially when the operational setup (equipment, callsign with /p) remains consistent across locations. And when using the phone, the system could automatically use the GPS location to determine the locator.  Of course, this would be an option, not a requirement. Users who prefer to create new stations for each location could continue to do so, while portable mode offers a streamlined alternative for those who need it.

HB9HIL, 2024-09-09:
@Beiri22 The problem is that the whole backend of Wavelog, it's complete structure is based on fixed locations. You can't make them "dynamically". If you import QSO's with Locator XY in a certain Location which has Location YZ configured, Your QSO's will be exported with Locator YZ. The locator which is defined in the station location. The station locations are also part of the authorization which is used to determine if a QSO is accessible by the user or if it belongs to another one. 

A "Portable-Mode" like you suggest it, would mean to completely rebuild Wavelogs Backend Structure, which is not planned and probably won't be ever. 

What I can tell you is how I manage this stuff as I operator /mobile for about 99.9%. 

For the most awards and QSL's (like LoTW) only a 4 char grid is calculated. My 4 char grid is `JN47`. So I added one location for my mobile operation in the big square `JN47` and I use this location wherever I operate from my truck in this area (which is arround 100km in middle europe). So only 1 location for every _generic_ mobile operation from grid JN47.

if I operate SOTA or POTA, I create new locations for it with the specific SOTA or POTA (or both) reference. This way I can make sure that the QSO's are seperated from the rest and are exported with the correct refs. 

So if you want to operator just _generic portable_ operation, create 1 station location for one big maidenhead. if you need any specific refs for SOTA, POTA, IOTA, WWFF or other SIG's you need to create a new station location.

HB9HIL, 2024-09-09:
PS: There is an option in your user menu called "Station Locations Quickswitch". If you enable this you can define certain locations as favourites in the station setup:

[image]

Aswell you have a new menu button on the upper right to quickly switch between those favourites

[image]

Beiri22, 2024-09-09:
Given how closely the management of stations is integrated with the entire data model and system, making changes would involve significant effort. However, the idea of implementing a larger grid for mobile use is definitely worth exploring. I will try this! Thank you!

closed 2024-09-13 by Beiri22