wavelog/wavelog #913: Automatically set or at least verify ITU and CQ zone from Locator

Beiri22, 2024-09-09:
Hi, 
I once came up with a similar request. Curently I do this manually with https://zone-check.eu/ -- maybe we could get to now, how they do it and include a similar functionality.

You could have a look at their scripts at first:
https://zone-check.eu/includes/locgrid.min.js
https://zone-check.eu/includes/zones.cq.js - there is some variable cqzones with i think google maps encoded paths.
https://zone-check.eu/includes/zones.itu.js -  there is some variable ituzones with i think google maps encoded paths.
https://zone-check.eu/includes/zones.min.js

maybe this could be an inspiration...

and sry to cause so much work.

AndreasK79, 2024-09-10:
Hi,

It is an interesting thought, but it is only useful in the map.
I reproduced the feature locally:
[image]

What it does, is that it uses the coordinates given from the map (on mouse move), and checks if those coordinates exists inside one of the zones, then returns the zone from there. The data from drawing the zones are used.

From what I understand, you want this in other parts of Wavelog?

Beiri22, 2024-09-10:
Hi Andreas,

You could use it, when the callbook does not provide an ITU and/or CQ zone but a locator (when creating a log or pressing the get from callbook button later). Locator->Coord->ITU/CQ.

Also you could use it to find mismatching information in logs. I thought there was some search mismatching cq zone feature right now; which could also find mismatching itu zones...

Beiri22, 2024-09-10:
Why manually? The checker could output results like a) wrong cq, b) missing cq, c) wrong itu, d) missing itu ... you make 4 buttons: Fill all missing (itu, cq), correct all wrong (itu, cq). e voila :-)

AndreasK79, 2024-09-10:
Give me a data source with all the information I need, and I will fix that.

AndreasK79, 2024-09-10:
You can probably determine the ITU zone from a grid, but what if the operation comes from a border (2 grids) or a corner (4 grids)? Which zone should we then take if that operation takes place from a border zone?

If we do not have the grid, it's a guessing game. Which ITU zone do you set in Canada, USA, Australia, China, Russia etc with multiple ITU zones?

Beiri22, 2024-09-10:
What we could do is, **given a locator**:
No-Border-No-Corner-Case: Checking and autofill possible.
Border or Corner case: At least we could check, if the given ITU and CQ zones would match the border or corner. We cannot fill without guessing. We could however tell the user that there is another category like edge cases where multiple ITU / CQ-Zones would be possible.

AndreasK79, 2024-09-10:
I quickly made an ITU zone checker by doing the same as I did with the CQ zone checker, and found that the data contained in the database was not complete, so the source needs to be updated to be able to determine which zones are wrong.
I quickly found that zone 1 and 4 are not set at all in the table, and would render Alaska and parts of Canada with a "wrong zone".

As for determining the edge cases, it is difficult to say. It all depends on your log, and probably what part of the world you are in.

I guess an ITU zone fixer would go like this (given that you have a complete list of DXCC and ITU zones):
1. Find zone by grid
2. Find by DXCC and take the default one (like we do with CQ zone).

Beiri22, 2024-09-10:
> and found that the data contained in the database was not complete
Do you mean the zone-check scripts or what database do you refer to?

AndreasK79, 2024-09-10:
The zone checker is based on a database table called dxcc_master. This table contains information about the dxcc.
We need this as a datasource to find incorrect zones.

Beiri22, 2024-09-10:
So this is a database table of the current version of wavelog. Are the variables cqzones and ituzones of zone-check.eu complete?

AndreasK79, 2024-09-10:
What do you mean? zone-check.eu uses longitude/latitude to find cq/itu zones. That is all well and good if we have a gridsquare, but it is not a database we can download and use.

Beiri22, 2024-09-10:
It may not be a database, but couldnt we extract the polygon information and make it a database?

AndreasK79, 2024-09-10:
We have that data. That is in use in the ITU award map and the advanced logbook map. Still, it would only be useful if you have a grid or longitude/latitude.

It is not useful for making an ITU zone checker. You need a list of all ITU zones for each DXCC to do that.

Beiri22, 2024-09-10:
Everything I am talking about is for the case that **the log has a locator ** (from direct entry or callbook lookup).

AndreasK79, 2024-09-10:
So all you wants is that in QSO entry when a grid is entered (manually/callbook), the CQ/ITU zones will be filled based on that?

Beiri22, 2024-09-10:
That was my main point. Lots of people do not enter the ITU zone in their qrz-profile, so this field is empty half the time.

Side point would be checking the plausibility of locator and itu/cq (as additional feature of the cq checker)

closed 2024-09-18 by AndreasK79