wavelog/wavelog #931: Bug Report: Misleading Note Content Information in QSO Note

besynnerlig, 2024-09-12:
#### **Describe the Bug**
When creating a QSO Note, the following text is displayed: 

> "Note content is used within Wavelog only and is not exported to other services."
[image]

However, this is misleading. After uploading the QSO to external services (like QRZ.COM), the note content is transferred as well. The issue arises because users expect the note to remain private to Wavelog, as per the message, but it is visible in external services' logbooks.

This behavior is causing confusion, as the current message suggests that the note remains within Wavelog.

#### **Version Information**
- **Wavelog**: Release 1.8.5

#### **Steps to Reproduce**
1. Go to `POST -> Post QSO`
2. Add a QSO and enter text in the "Notes" tab.
3. Upload the QSO to QRZ.COM.
4. Go to QRZ.COM, edit the QSO, and observe that the note content is present in the log.

#### **Expected Behavior**
Either:
- The note should not be transferred to external services like QRZ.COM as the message suggests, **OR**
- The text in the application should be updated to accurately reflect that the note **will** be shared with external services when the QSO is uploaded.

#### **Screenshots**
Below is an example where the note content appears in the QRZ.COM logbook:
[image]

HB9HIL, 2024-09-12:
Hello @besynnerlig 

You are absolutely right! This should not be. 

this is getting fixed. Thank you for the report!

HB9HIL, 2024-09-15:
fixed

closed 2024-09-15 by HB9HIL