wavelog/wavelog #932: Feature Request: Add Button to Report New Spot to Cluster Server

besynnerlig, 2024-09-12:
### **Feature Request: Add Button to Report New Spot to Cluster Server**

#### **What Feature is Missing?**
It would be a great enhancement to add a button on the page for adding a QSO. After entering a callsign, users could report a new spot to the cluster server with just one click. The feature would use the newly entered callsign, frequency, and traffic mode to automate the spot reporting.

#### **Why is This Feature Important?**
This feature would streamline the process of reporting new spots to the cluster server, saving time and reducing manual input. By integrating this directly into the QSO logging process, users can quickly contribute spots while logging their QSOs, enhancing the overall user experience and workflow.

#### **Is This Feature Personal or Beneficial to Others as Well?**
This feature would benefit the entire Wavelog community, particularly those who regularly report spots to the cluster server. It adds a valuable layer of efficiency for all users who engage in both logging and spotting, making the software more versatile and user-friendly.

int2001, 2024-09-12:
Wavelog is a tool which runs on your websever (or at a hoster) and provides a logbook (and cluster) to many users.
Most DXClusters do not have any kind of authentication. You simply login via Telnet and post your spots.

All on a trustlevel with only a Callsign.

if we put a spot-function into wavelog, all spots would come from the IP of your WL-Server.
That means: If someone is doing evil things (e.g. spotting everyone he sees on FT8) with that, your instance/IP will be banned very quickly on the Cluster.

This is (one of) the main reason(s) why we decided not to implement a spotting-function within WL.

Maybe this changes one day, when clusters are more secure. But currently it is readonly.

closed 2024-09-12 by int2001

besynnerlig, 2024-09-13:
Thank you for the detailed explanation regarding the concerns about implementing a spotting function in Wavelog. I understand that the primary issue is not necessarily that sharing a server increases the likelihood of abuse, but rather that it heightens the risk of the server being blacklisted due to the actions of a single user.

That said, there might be alternative approaches to mitigate this risk. For example, some platforms provide ways to forward spots directly from the user’s browser, which would allow the spot to originate from the user's own IP address rather than the server. This could potentially remove the server from acting as an intermediary for spotting, reducing the risk for shared installations. While it would require further exploration to see if such an approach is widely supported by clusters, it could be an option to consider.

Additionally, if someone intended to abuse the PacketCluster network, they would likely find more direct and easier ways than using a logging program. I'm not sure if this kind of abuse is as prevalent as it might seem.

In any case, I appreciate the careful consideration of these potential issues, and perhaps there will be future solutions as cluster technology evolves. Thanks again for all the work you put into Wavelog!

int2001, 2024-09-13:
Happy to help.
Would be very nice if you can share informations about a Cluster which provides a https-API for End-Users.

EA1DDO, 2025-07-06:
> Maybe this changes one day, when clusters are more secure. But currently it is readonly.

That may apply for HF, however, the ea4ure.com:7300 cluster needs to register first to get the password, to be able to login.
That for HF use.
But for VHF things are different. In order to keep working services like DXmaps.com we use to spot every contact we complete. Log software has the option to automatically spot every valid QSO.

Right now I am participating on the CQWWVHF, using first time Wavelog, and my contacts are missing in DXMaps. If spot option is not included in future releases... Wavelog will loose interest, at least for VHF use.

Thank you
73, Maximo EA1DDO

[image]

int2001, 2025-07-06:
Sorry, we discussed that often internally. a feature for spotting isn't planned at main-wavelog

you're free to extend it for your uses.

EA1DDO, 2025-07-06:
> you're free to extend it for your uses.

That could be a good solution, to provide an extra module to anyone interested could install it.

int2001, 2025-07-06:
great idea. we appreciate contribution.