wavelog/wavelog #945: Clublog-Sync items in the 3rd party services

Beiri22, 2024-09-13:
Hi,

I see that there are cronjobs to sync with clublog, but the menu items in 3rd party services (german: Drittanbieter-Dienste) is missing. I know, this is because the feature only would show a blank page and not a nice gui. But I personally would prefer to see the link there. Maybe labeled as Clublog (alpha phase) or however.

int2001, 2024-09-14:
Ok, let's try to switch to german in this case. i'll translate afterwards for all other participants.
we didn't get the point here.

German:
Lass es uns auf Deutsch versuchen. Übersetze anschliessend für alle anderen hier auf Englisch.
Kannst Du bitte noch mal das Problem genau beschreiben? 

Ich versuche mal zu interpretieren: Für Clublog gibt es aktuell keine Seite auf der man den Sync per Hand anstossen kann. Du möchtest gerne eine solche Seite haben. Richtig?

Falls es das ist: Wir haben sehr bewusst auf "cron-only" gesetzt, da die automatische Synchronisierung im Hintergrund fehlerfreier läuft, als der manuelle Trigger.

Beiri22, 2024-09-14:
Du verstehst es richtig. Die Seite für das manuelle Anstoßen des Clublog-Up/Downloads fehlt. Nun weiß ich, dass es schwierig ist, die ganze Oberfläche da neu aus dem Boden zu stampfen. Für mich würde es schon ausreichen, wenn eine einfache Seite käme, erstmal als Alpha-Version mit zwei Buttons Up- und Download, die jeweils einfach die URL aufrufen, mit denen die Funktion jetzt schon erreichbar ist. 

Aus purer Neugier: Wie kann der automatische Cronjob besser laufen als die manuelle Routine? Sind die beiden unterschiedlich implementiert?

int2001, 2024-09-15:
Ich befriedige mal zuerst die Neugier ;) - Daraus ergibt sich nämlich der Grund:
Bei einem Log, welches nur von Dir genutzt wird (Einzeluser), ist es relativ egal wann/wie Du den Sync zu Clublog (oder auch eQSL/LoTW/QRZ) anstösst. Sobald jedoch mehrere Nutzer Wavelog nutzen kann es zu sogenannten "überholeffekten" und auch Timeouts kommen. Dies wird durch die synchronisierung im Hintergrund ausgeschlossen.
Der cronjob sorgt dafür, dass Station für Station sauber und geordnet synchronisiert wird. Folge: Es kommt zu weniger Fehlern.
Und bevor die Frage aufkommt: Das betrifft auch die anderen Dienste. Daher gibt es ein config-setting, dass den manuellen Sync für die anderen Dienste ebenfalls deaktiviert. Prinzipiell ist der geordnete Aufruf per cron immer dem manuellen anstuppsen vorzuziehen.

Hast Du auf Deiner Instanz nicht die Möglichkeit crons laufen zu lassen? Du kannst die Sync-Jobs ja über den cronmanager sogar selber "takten" und dort einstellen wann was gesynct werden soll.

Beiri22, 2024-09-15:
Ich glaube, ich verstehe die Problematik. Da es beim manuellen auslösen nutzerübergreifend kein Pooling oder ähnliches gibt, könnten zeitgleiche Zugriffe auf die APIs zurückgewiesen werden. Wird mich als Einzelnutzer nicht betreffen, aber ich schaue Mal, wie das mit dem Cron Jobs aussieht. Meine Installation läuft auf Docker. Zumindest in einem anderen dicker Container habe ich Cron Jobs Mal eingerichtet.
Als Anfänger habe ich das Gefühl, das ein wenig Kontrolle übernehmen den Zeitpunkt des sync ganz gut ist. Damit ich dann weiß, dass meine qsos wirklich in Ordnung sind, wenn sie synchronisiert werden. Aber vielleicht geht dieses Gefühl mit der Zeit weg. Aber wir ist denn langfristig der Plan ....?
1 soll manuelle Synchronisierung ein poolingsystem bekommen und dann quasi der Nutzer sehen, ob er gerade in der Warteschlange ist oder schon seine Daten synchronisiert werden?
2 wird die manuelle Synchronisierung gänzlich wegfallen?

int2001, 2024-09-15:
Unsere Standard-docker-installation hat bereits vorgefertigte Cronjobs, die feuern. Wenn Du Dir allerdings selbst einen Container gebaut hast, können wir da wenig supporten.

idR musst Du bei unserem Docker-Image nichts ändern.

Langfristig: Ggf. eine Übersicht über die letzten gesyncten Kontakte. Aber das ist Zukunftsmusik.

Ob die manuelle Sync wegfällt: denke eher nicht. es gibt kleine und grosse installationen...