wavelog/wavelog #952: Mode not always selected/shown in live-qso when using wavelog-gate

Nailcoding, 2024-09-15:
When using wavelog-gate to get trx data in the live-qso area, the mode is very often not shown (empty) in the dropdown. While band works always, the mode does not work in more than 50% of the mode changes. It is shown correctly in wavelog-gate, but not in wavelog. This maybe also a wavelog-gate bug. 
Versions:
Wavelog: 1.8.2
Wavelog-gate: 1.0.16

int2001, 2024-09-15:
Can u please share a screenshot?

Nailcoding, 2024-09-16:
[image]

int2001, 2024-09-16:
thank u.
the screenshot from WLGate and the expected value is missing.

Could you pse provide the following Informations:
- Screenshot WLGate (Tab "Status")
- Information about your TRX
- Given Mode vom TRX which fails.

tu

Nailcoding, 2024-09-16:
[image]

Nailcoding, 2024-09-16:
[image]

Nailcoding, 2024-09-16:
[image]

int2001, 2024-09-17:
okay.
can you pse check if you've (accidently) deactivated the submodes for SSB (USB/LSB) at the Adminmenu?

go to the Menue with the gear-symbol, then to Modes and search for SSB (upper right).
The Submodes USB/LSB should not be deactivated. If they're activated, pse send a screenshot from that as well.

[image]

int2001, 2024-09-17:
One more: due to trafficsaving WLGate only updates Wavelog of you QSY (change the QRG with the VFO). So it may be, that a mode-change only isn't detected to sync.

That'd be a bug indeed.

Nailcoding, 2024-09-17:
The deactivated submodes were the reason for the problem. I did not accidentally deactivate them. I did this to keep the dropdown list as short as possible. I always only enter SSB, not USB or LSB.

closed 2024-09-19 by Nailcoding