wavelog/wavelog #953: Adif Export function for search results in advanced logbook

Beiri22, 2024-09-15:
I would like to prefer an option to export the search results in advanced logbook. There I can select all e.g. not already sent qsl cards of a specific date range.

AndreasK79, 2024-09-15:
Did you click the button for ADIF?

Beiri22, 2024-09-15:
oh. This is misleading.

On the left side I expected the filter criteria (like quick filter, qsl-filter, and filter); but the actions-button should be on the right side. I totally overlooked that. Especially as there are destinct actions on the right side that are not under the action button - like Edit, Map. 

My brain works like
left define filter criteria, click search, number of results, do something with the results

not
define filter criteria and do something with the results, number of results, another filter criterion (station), click filter, some other actions

Beiri22, 2024-09-15:
(plus, does duplicates search globally or only duplicates on the filtered?, same vor invalid _ i mean, both could be criteria on the left side...)

HB9HIL, 2024-09-15:
to be honest... I don't see any issue here..

Beiri22, 2024-09-15:
So you don't see any advantage in a clear and ordered user interface? Actions should not be mixed with search criteria. I mean, moving some buttons is done in a few minutes. Taking about improving usability might take a little longer. I'm open to talk about it...

int2001, 2024-09-16:
no issues visible as well here.
it's personal flavour. Menu exists for a long time as it is. No complains so far and also no plans to change that because of one user. Sorry.

73

closed 2024-09-29 by github-actions[bot]