wavelog/wavelog #971: Missing Counties in drop down box when creating a station location

nq0m, 2024-09-18:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

At least for Kansas, not all counties show up in the drop down box - I suspect there's a limit somewhere in the number of items available in the drop down box.  Kansas has 105 counties total, and the last 4 alphabetically are not shown:

Wichita
Woodson
Wilson
Wyandotte

Wyandotte is one of the most heavily populated counties in Kansas, so I'm surprised this has never been noticed before.

**Steps to Reproduce**
1. Go to Create Station Location
2. Select United States, and the state of Kansas
3. In the drop list for counties, the last one shown is Washington, 4 are missing

HB9HIL, 2024-09-18:
thx for reporting this :) fixed soon and will be in the next release

HB9HIL, 2024-09-19:
fixed

PR #973 (2024-09-19): Fix number of US counties
fixes the available numbers of US counties in the dropdowns. 

Texas has 254 counties and with this the most counties of the US states. Therefore we have to show the all. 

https://en.wikipedia.org/wiki/County_statistics_of_the_United_States

`fixes #971`
closed 2024-09-19 by HB9HIL