wavelog/wavelog #977: Add Support for Prefilling Callsign via URL Parameters in Logging Window

besynnerlig, 2024-09-19:
**What Feature is Missing?**

I would like to request the addition of a feature in Wavelog that allows the logging window to accept query parameters in the URL to prefill form fields, specifically the callsign. This means that when accessing a URL like:

```
https://wavelog.example.com/index.php/qso?callsign=SE6S
```

Wavelog would open the logging window with the callsign field already populated with "SE6S". This functionality would enable external applications or scripts to initiate the logging process with prefilled data, enhancing interoperability.

**Why is This Feature Important?**

I use an SDR radio (SmartSDR connected to a FlexRadio) that displays spots on its panadapter. When I click on one of these spots, my setup can be configured to automatically open a URL in my browser. If Wavelog could accept the callsign as a query parameter, it would streamline the process of start logging a QSO initiated from my radio. This integration would make starting a QSO by clicking a spot in the panadapter quick and seamless, significantly improving my operational efficiency.

Moreover, this feature would reduce manual data entry errors and save time.

**Is This Feature Personal or Beneficial to Others as Well?**

While this feature addresses my specific use case, I believe it would be beneficial to many other Wavelog users. Operators who use SDR radios or other equipment that can trigger browser actions would find this integration valuable. It would enable them to:

- **Enhance Efficiency**: Quickly start logging QSOs with prefilled information, reducing the steps required.
- **Improve Accuracy**: Minimize manual data entry, reducing the likelihood of errors.
- **Increase Flexibility**: Allow for better integration with various hardware and software setups, catering to diverse operating styles.

Implementing this feature would make Wavelog more versatile and appealing to a broader user base, encouraging adoption and satisfaction among amateur radio operators who seek seamless integration between their equipment and logging software.

[image]
_Spots appearing on the SmartSDR Panadapter_

phl0, 2024-09-19:
Is there documentation about this feature? For example parameter names etc.?

besynnerlig, 2024-09-19:
Thanks @HB9HIL 
Sounds great and I will test it tomorrow and let you know the results!

HB9HIL, 2024-10-05:
Implemented

besynnerlig, 2024-10-06:
Dear Wavelog Development Team,

I would like to extend my sincere thanks to you all, (especially to  @HB9HIL) for your excellent work on implementing the feature request. It is clear that you have been responsive to user's needs, and I truly appreciate your attention to detail in making this improvement.

This new API will undoubtedly prove to be incredibly useful, especially for those of us integrating Wavelog with SDR setups like FlexRadio. It will enhance the overall experience, improving both efficiency and accuracy in logging QSOs.

Thank you once again for your dedication to making Wavelog better for the entire community. Your hard work does not go unnoticed!

closed 2024-10-06 by besynnerlig