wavelog/wavelog #981: Add Lightbox Feature for Viewing Full-Size QRZ.com Profile Images on Add QSO Page

besynnerlig, 2024-09-20:
**What Feature is Missing?**

I believe the ability to view a QRZ.com profile image in full size when adding a QSO would be great. Currently, the profile image displayed on the Add QSO page is a bit small, which is understandable due to the limited space on the webpage. However, it would be useful if users could click on the image to enlarge it, without needing to right-click and open it in a new tab. A "lightbox" effect would allow users to click on the image to view it in its full resolution within the same page, and then easily dismiss the enlarged view by clicking outside the image or pressing the escape key.

This can be achieved by implementing a lightbox-style functionality, where the thumbnail is clickable, and an overlay with the full-size image appears, providing a seamless and user-friendly experience. This approach is used widely in modern web applications for displaying images and offers an elegant way to avoid opening new browser tabs.

**Why is This Feature Important?**

This feature would greatly improve the usability and user experience of Wavelog, particularly for users who wish to see QRZ.com profile images in greater detail when managing their QSO entries. Instead of going through the extra steps of right-clicking and opening a new tab for a larger view, users would have a more convenient, intuitive way to quickly view and dismiss the full-size image. 

In addition, this enhances the visual presentation of Wavelog and makes it more user-friendly for hams who value efficiency when managing QSOs.

**Is This Feature Personal or Beneficial to Others as Well?**

This feature would be beneficial to the broader Wavelog community. Many users likely want to see full-size QRZ.com profile images without the hassle of opening a new tab, which interrupts the flow of adding or editing QSOs. The lightbox functionality is a common and familiar pattern used across the web, and implementing it in Wavelog would enhance the experience for all users who manage QSOs and reference QRZ.com profiles during logging.

int2001, 2024-09-20:
to make sure i understand your feature request correct: you like to get a "enlarge" button on the image??

Have you seen the little "globe" with the QRZ-Logo next to the callsign (See screenshot)?
A single click (left mouse key) on it opens the whole qrz-page of the user - in a new Tab. Very convenient, because there's much more information than only a picture. 

[image]

besynnerlig, 2024-09-20:
Thanks for your response. No, I'm not suggesting an "enlarge" button. The functionality I’m referring to is demonstrated at the [Lightbox2 Demo Page](https://lokeshdhakar.com/projects/lightbox2/). It allows the image to quickly open in full scale and then close just as easily without needing to leave the Wavelog page.

Yes, I’ve seen the globe icon and use it frequently to look up QRZ.com info. However, that process involves a click to open the QRZ.com page in a new tab, another click to enlarge the profile image, and then two more clicks to return to Wavelog. In a CW QSO, I don’t have time to go through all of that.

The QSO logging page is a perfect use case where lightbox excels, providing a quick and easy way for the user to view the profile image without multiple steps.

int2001, 2024-09-20:
ok. tbh: it's kind of "enlarging" but with cosmetics.

imho nothing urgent, just a little adjustment to make Wavelog look a bit more shiny.
tnx for the idea.

HB9HIL, 2024-09-22:
@besynnerlig 

See the linked PR which solves your request :)

besynnerlig, 2024-09-23:
> @besynnerlig
> 
> See the linked PR which solves your request :)

Thank you so much for the quick action and the PR! I really appreciate how swiftly you addressed the request. It’s great to see how responsive the Wavelog team is, and I’m excited to see the feature in action. Your effort in making this improvement is truly appreciated!

int2001, 2024-09-23:
Spread the word ;)

So we (or u) can close this issue?!

besynnerlig, 2024-09-23:
Yes! Thanks!

closed 2024-09-23 by besynnerlig