wavelog/wavelog #983: API callsign lookup function always returns workedBefore: false,

besynnerlig, 2024-09-20:
The API callsign lookup function in application/controllers/Api.php always returns workedBefore: false, even for callsigns that have been worked before. The expectation is that the field should accurately reflect whether the callsign has been previously logged as a QSO.

**Version Information**

    OS: Debian 12
    PHP: 8.2.20
    MySQL/MariaDB: 10.11.6-MariaDB-0+deb12u1
    Wavelog: 1.8.5

**Steps to Reproduce**

1. Use a third-party app to post a callsign lookup request to the API.
2. Inspect the returned result.
3. Notice that workedBefore is always false, even for callsigns that have been worked before.

Expected Behavior workedBefore should return true for callsigns that have been worked before and are present in the log.

Screenshots
[image]

Desktop Information:

    OS: Ubuntu 22.04
    Browser: Firefox
    Version: 117.0.1

Smartphone Information: N/A

**Additional Context**

This bug affects the callsign lookup API, where workedBefore always returns false regardless of whether the callsign is in the logbook or not. This behavior creates issues for external services relying on accurate lookup results.

int2001, 2024-09-20:
Which API are you calling?
What does our Wiki/Documentation say about the API-Endpoint you're using?

besynnerlig, 2024-09-20:
Dear Joerg, you edited your reply before I posted my answer, but here it is.

Thank you so much for your quick response and for pointing me in the right direction.

To clarify, I am performing a DXCC lookup for a callsign using the same API that WaveLogGate relies on, located at https://wavelog.example.com/index.php/api/lookup. I assumed this was the appropriate endpoint, as it's being used successfully by other applications in the same ecosystem.

Regarding the Wiki and documentation, I did check the page at https://github.com/wavelog/wavelog/wiki/API but couldn’t find any specific information about this particular API endpoint. I realize that sometimes documentation can lag behind the codebase, which is perfectly understandable in an active project like this. That said, if I missed something or used an undocumented part of the system, please let me know. My intention is certainly not to overstep or misuse any functionality—just to be helpful and contribute where I can.

I appreciate your guidance, and I’m happy to make any changes or adjustments as needed. I’m definitely not here to complain, and I’d love to continue assisting in any way that benefits the project.

Thank you again for your support and for maintaining such a valuable tool!

Warm regards

int2001, 2024-09-20:
I like the answer :) 
The current API is really only used internally. Maybe we change that one day and document the official use there.

It can be very helpful for us to understand your UseCase a bit more. That can enable us to provide the right information you need for the external applications you mentioned.

What i try to say: if there's a third-party application which uses the API already (besides our own), we can make it official. It depends on the type of App.

Best 73 and thanks for the open words, DJ7NT

besynnerlig, 2024-09-20:
Dear Joerg,

Thank you for your kind words and for being open to this discussion.

To provide more insight into my use case, the third-party app I’m developing is designed to integrate WaveLog with FlexRadio’s SmartSDR panadapter. The app subscribes to DX spots from a DXCluster server, performs DXCC lookups using the API, and sends the spots to be displayed on the FlexRadio panadapter. The integration highlights needed DXCCs with custom colors and provides additional information, like whether the station is a LoTW member, to enhance the experience.

The workedBefore information is crucial for this workflow, as it allows operators to quickly identify whether they’ve already worked a particular callsign. This helps them focus on new contacts and avoid unnecessary duplicates, which is especially useful for those chasing DXCC awards. Displaying this information directly on the SmartSDR panadapter makes it easy for operators to see which stations they still need without manually searching through their logs.

Additionally, when a spot is clicked on the panadapter, the WaveLog logging page opens with the clicked spot’s callsign prefilled, making the logging process seamless and efficient. This integration creates a very smooth and pleasant user experience, as operators can immediately start logging QSOs without any manual input.

The app’s primary goal is to bridge logging and DX hunting with real-time visualization on the FlexRadio. I would be thrilled if this use case aligns with the goals of the WaveLog project, and I would love to collaborate on making the API more suitable for external integrations like this.

Best 73
[image]

int2001, 2024-09-20:
Wow. This is a thing we'll definitely going to support ;)
I like.

I think we've to align on some payload. The current lookup won't help here and tbh: i can't adjust it, because our own DXClusterAPI relies on it.

proposal would be a new API-Endpoint for that.
You'll need:
- key (of course)
- station_id (optional. if not given every station_id of the user will be searched)
- BAND (This is important, because it wont help if a station is worked via "DMR" but needed on 40m)
- Mode (same as band)

the result of the API could be nearly the same, but i'd like to add the following informations:
- dxcc_worked_on_band_mode (boolean)
- dxcc_worked (boolean)
- call_worked_on_band_mode (boolean)
- call_worked (boolean)

with that information you should be able to display the right information on the flex.

let me know what you think about the proposal.

Best 73 DJ7NT

besynnerlig, 2024-09-22:
Dear Joerg,

Thank you so much for your enthusiasm and for the proposal. I'm thrilled to hear that this is something you'll support!

At the moment, I’m still in the process of testing various parts of my application and evaluating the results. I’ve attached an image to give you an idea of how the SmartSDR panadapter currently displays spots for callsigns with different colors, which are fully configurable based on whether the callsign has been worked before or logs to LoTW. It’s been exciting to see everything come together, and your proposal for a new API endpoint sounds like exactly what’s needed to push this integration forward.

I appreciate your suggestion to add key, station_id, band, and mode filters, along with the additional dxcc_worked and call_worked fields. This approach would definitely help me display the right information on the FlexRadio, ensuring operators can quickly identify which stations they’ve worked before, both in general and on specific bands/modes.

I’ll continue testing and piecing together the different parts of the app over the next few days. Once I have more insights from my testing, I’ll return here with any further thoughts or suggestions for making the API even more suitable for external integrations like this.

Thanks again for your support, and I look forward to working together on this!

Best 73

[image]
(The "Worked before" status is currently simulated and randomized. It’s shown with a transparent background in this image to suggest that these callsigns have already been worked.)

int2001, 2024-09-22:
ok, to make sure i understood it correctly.
You're fine with the proposal?

if yes: i am going to create a PR with that feature and you can test?

besynnerlig, 2024-09-22:
Dear Joerg,

Thank you again for the proposal I’m happy with it overall!

One thing I’d like to mention is that the lotw_member field is quite important for the integration. As in the current internal API, it can either be false or a number, with the number providing additional information about when LoTW was last updated. Including this would be very useful.

Additionally, to make the API even more versatile, could you consider adding the following fields as well?

- dxcc_confirmed_on_band (boolean)
- dxcc_confirmed_on_band_mode (boolean)
- dxcc_confirmed (boolean)

These would further enhance the functionality, allowing users to easily see if a DXCC belonging to a callsign has been confirmed on a specific band/mode.

Thank you again for your effort on this, and I’m excited to test the feature when it’s ready so please proceed with the PR.

Best 73

int2001, 2024-09-30:
Hi @besynnerlig,

i am going to start with the implementation during this week.
But before i do that, pse be sure and confirm(!) that your API-Call can provide the following informations (those are mandatory for your request):

* key (of course)
* station_id (optional. if not given every station_id of the user will be searched)
* BAND (This is important, because it wont help if a station is worked via "DMR" but needed on 40m)
* Mode (same as band)

Regarding the QSL (you proposed LoTW only) we'd like to take the Usersetting for QSL-Methods (See account-setting). This makes it more flexible than only one service.

besynnerlig, 2024-09-30:
Hi @int2001,

Yes, I can provide all the information you requested in the payload sent to the Wavelog API. In addition to what you specified, you'll need the callsign that Wavelog will use to provide augmented data (which in my case is the spotted callsign) in the payload.

Regarding `station_id`, wouldn't it be more logical to send the `Station Logbook name` (the active logbook name, or better yet, its ID, which is currently not shown in the UI) since it's used for analytics throughout Wavelog? This would maintain consistency. If not set, the currently active logbook could be used as the default.

As for determining which authority (QSL method) should be used to identify whether a DXCC is needed, I earlier suggested LoTW because, as far as I know, ARRL is the only authority that grants DXCC credit for a QSO if I want to apply for a DXCC award. However, you raise a good point, many radio amateurs might not be interested in applying for awards and could consider a DXCC to be "worked" even if confirmed through other QSL services. Personally, I've set my default QSL methods in Wavelog to QSL, LoTW, eQSL, QRZ.com, and Clublog because I use all these services for sending QSLs, and I like to see information from them in the Wavelog logbook overview. However, in Wavelog -> Awards -> DXCC, I've set "Show QSO with QSL Type" to only show QSL and LoTW.

Could I suggest that this new API uses that setting for determining if a DXCC is needed or not?

besynnerlig, 2024-09-30:
Thanks, I will do that

int2001, 2024-10-04:
Closing this one here. is fixed with #1017
tnx for the input and building the Flex-Plugin