wavelog/wavelog #987: QSO-field for confirmed long path contact?

Beiri22, 2024-09-21:
Hi,

I just wonder if there could be a check box (create qso, edit qso) for usage of long path -- which would result in a different distance calculation and make a long path drawn on the globe map...

HB9HIL, 2024-09-22:
It's totally legit to ask for that as `ANT_PATH` is an official ADIF field. We currently have no possibility to edit that. I plan to do some frontend stuff for the QSO edit field. I will address this aswell then, okay?

I currently have a lot of private stuff to do (Moving Home from A to B). After that I will take care of that. Please be patient.

Beiri22, 2024-09-22:
i will be

HB9HIL, 2024-11-05:
@Beiri22 

First step is done. You now can edit ANT_PATH in the QSO edit. The path drawings and distance calculation will be addressed in another PR.

HB9HIL, 2024-12-02:
Distance calculation is also included. The path drawings will be adressed some other day as they are really complex. The main request is solved so I guess we can close this issue.