wavelog/wavelog #998: Some field filled wrong from callbook.

Beiri22, 2024-09-24:
I am using the live qso input. At first I note the callsign of the cq calling station, then I call. As portable station not all my calls come through. So from time to time I will overwrite the callsign because my first try was not successfull. This does sometimes, irregularly result in incorrect filled fields by callbook (like qsl via). I don't know, how the system works. Maybe the fields are not reset (correctly) when entering a new callsign? Could you check this?

dg9vh, 2024-09-24:
How do you procede in detail? Do you leave the callsign-field with TAB or do you just enter the new callsign without leaving the field?

Beiri22, 2024-09-24:
My station is portable, so I use an android smartphone. There is no tab key. After entering the callsign, I click into some other field, so that I can check if the callsign was found. It might be, that in some cases when the contact signals 59 I do not click into another field. But these should be rare cases, because I really like to check if the callsign exists in qrz.com in the first row.

If there was a button like reset, I would use this. Especially if this button would also reset the start time to the automatic mode. But there is none. Sometimes I just reload the page in case of negative contact to reset everything; but especially when changing band, then the "old" band is prefilled. This is not convenient.

dg9vh, 2024-09-24:
I can't check it on android actually but on PC-Browser-versions there is a reset-button at the bottom of the log-window:
[image]
Can't imagine, this would be the same on android...

Beiri22, 2024-09-24:
You're right. This was totally out of my sight for 2 reasons:
- it is on the lower end of the page, that is not visible when editing in the upper region. I would expect this button next to the callsign input, as it resets the callsign-related data
- using dark mode, the grey of the button and the grey background have very little contrast.

But nevertheless there could be a problem when importing the callbook data.

dg9vh, 2024-09-24:
Well, right.. Maybe it also could be an idea for the dev-team to have this buttons (save and reset) on top of the page when using mobile-devices...
maybe the position could be changed via CSS as an responsive thing.. I don't know.

Beiri22, 2024-09-24:
I mean, the save button on the bottom seams ok. You scroll down while you enter the values; just the reset feels out of position. (You have a mini-reset button for the start timer inside the input field, this could also be an idea for the reset button; but it not only resets the callsign and callbook data, so maybe it should be separate) Dev team could talk about it.

spREGR, 2024-09-25:
You may want to have a look at CloudLog Offline app. It works well on a phone. Since, there is an iPhone app, I am sure the Android one exists as well 

You log your QSOs on a mobile optimized form and then export the data to your WaveLog instance via API. 
Hope this helps

73 VU2TUM

Beiri22, 2024-09-26:
If the app is offline I would miss the callbook reading features. Also you do not see previous contacts. I really like an integrated solution.

closed 2024-10-04 by int2001