wavelog/wavelog #1004: Problem Upload Hamlog (Hamoffice 6.0)

dk9ms, 2024-09-26:
Hi OM's
i write you in English, i hope you can interprete my problem. 
I use Hamoffice 6.0 . I tried to upload my complete Log ( more than 34000 Logs) at Wavelog. My Log began from 2002. After several tries to upload my complete Log ( Data amount around 13, 1 Mb) every Logdate began from 2002 and ended exactly at 17.06.2011. All logs before this date were visible, all other logs are supressed, After uploading the log, the info occurs that uploading process was ok.
Secondary i have delete my log and try to upload a little amount of my log beginning from 18.06.2011 to 17.06.2012. After that,
i saw 
I do not have any idea what the reason is, and what i have to do now.
Any help very appreciated! Many thanks for support
73 de Markus, DK9MS

phl0, 2024-09-26:
Did you see an error message? Are you using an own instance or the DARC DCL one?

dk9ms, 2024-09-27:
Hi Florian, tnx for answer. Problem has been solved!
Ich antworte mal in Deutsch:
Das Problem hat sich erledigt, ich habe mein Hamoffice Log (2002 bis 2024)
in einzelne "Jahres-Häppchen" zerlegt und jedes Jahres-Log einzeln
hochgeladen.
Dies hat funktioniert! Wenn man das gesamte Log hochgeladen hatte, war eine
Erfolgsmeldung da, aber jedesmal war beim Log bei 2011 Schluss!
Danke für die Rückantwort!
73 de Markus, DK9MS

phl0, 2024-09-28:
Eigentlich sollte es auch mit dem gesamten Log klappen. Gab es bei den einzelnen Dateien irgendwo einen Fehler? Ich würde einen/den Bug gerne ausräumen, sofern es einer ist. 😇

int2001, 2024-10-04:
Sollte mit #1029  gelöst sein. Das Log hast Du mir ja zugesendet,  Markus.
Können wir hier schliessen?

closed 2024-10-13 by github-actions[bot]