wavelog/wavelog #1005: Fresh installation fails on 'Could not migrate database. Ajax crashed.'

pa8s, 2024-09-26:
**Describe the bug**
Performing a fresh installation, the final step fails on 'Could not migrate database. Ajax crashed.'

**Version Information**
- OS: FreeBSD 13.1
- PHP:  8.2
- MariaDB:  10.11.8-1
- Wavelog:  1.8.5

**Steps to Reproduce**
1. Perform steps until 'Install Now'

**Expected Behavior**
Successful installation

**Screenshot**

[image]

**Logfile**

```
Wavelog Installer Debug Log
-------
Log Location: ../application/logs/installer_debug.log

DEBUG - 2024-09-26 19:54:43 --> Start writing config.php
DEBUG - 2024-09-26 19:54:44 --> File: config.php successfully written
DEBUG - 2024-09-26 19:54:44 --> Start writing database.php
DEBUG - 2024-09-26 19:54:45 --> File: database.php successfully written
DEBUG - 2024-09-26 19:54:45 --> Start creating database structure with assets/install.sql
ERROR - 2024-09-26 19:54:46 --> Creating database tables failed. Ajax crashed.
```

**Additional information**

Output for database user:

```
wavelog@localhost [(none)]> show databases;
+--------------------+
| Database           |
+--------------------+
| information_schema |
| wavelog            |
+--------------------+
2 rows in set (0.000 sec)

wavelog@localhost [(none)]> show tables in wavelog;
+-------------------------------+
| Tables_in_wavelog             |
+-------------------------------+
| TABLE_HRD_CONTACTS_V01        |
| api                           |
| cat                           |
| config                        |
| dxcc_entities                 |
| dxcc_exceptions               |
| dxcc_prefixes                 |
| migrations                    |
| notes                         |
| options                       |
| station_logbooks              |
| station_logbooks_relationship |
| station_profile               |
| timezones                     |
| users                         |
+-------------------------------+
15 rows in set (0.000 sec)

wavelog@localhost [(none)]> show privileges;
+--------------------------+---------------------------------------+--------------------------------------------------------------------+
| Privilege                | Context                               | Comment                                                            |
+--------------------------+---------------------------------------+--------------------------------------------------------------------+
| Alter                    | Tables                                | To alter the table                                                 |
| Alter routine            | Functions,Procedures                  | To alter or drop stored functions/procedures                       |
| Create                   | Databases,Tables,Indexes              | To create new databases and tables                                 |
| Create routine           | Databases                             | To use CREATE FUNCTION/PROCEDURE                                   |
| Create temporary tables  | Databases                             | To use CREATE TEMPORARY TABLE                                      |
| Create view              | Tables                                | To create new views                                                |
| Create user              | Server Admin                          | To create new users                                                |
| Delete                   | Tables                                | To delete existing rows                                            |
| Delete history           | Tables                                | To delete versioning table historical rows                         |
| Drop                     | Databases,Tables                      | To drop databases, tables, and views                               |
| Event                    | Server Admin                          | To create, alter, drop and execute events                          |
| Execute                  | Functions,Procedures                  | To execute stored routines                                         |
| File                     | File access on server                 | To read and write files on the server                              |
| Grant option             | Databases,Tables,Functions,Procedures | To give to other users those privileges you possess                |
| Index                    | Tables                                | To create or drop indexes                                          |
| Insert                   | Tables                                | To insert data into tables                                         |
| Lock tables              | Databases                             | To use LOCK TABLES (together with SELECT privilege)                |
| Process                  | Server Admin                          | To view the plain text of currently executing queries              |
| Proxy                    | Server Admin                          | To make proxy user possible                                        |
| References               | Databases,Tables                      | To have references on tables                                       |
| Reload                   | Server Admin                          | To reload or refresh tables, logs and privileges                   |
| Binlog admin             | Server                                | To purge binary logs                                               |
| Binlog monitor           | Server                                | To use SHOW BINLOG STATUS and SHOW BINARY LOG                      |
| Binlog replay            | Server                                | To use BINLOG (generated by mariadb-binlog)                        |
| Replication master admin | Server                                | To monitor connected slaves                                        |
| Replication slave admin  | Server                                | To start/stop slave and apply binlog events                        |
| Slave monitor            | Server                                | To use SHOW SLAVE STATUS and SHOW RELAYLOG EVENTS                  |
| Replication slave        | Server Admin                          | To read binary log events from the master                          |
| Select                   | Tables                                | To retrieve rows from table                                        |
| Show databases           | Server Admin                          | To see all databases with SHOW DATABASES                           |
| Show view                | Tables                                | To see views with SHOW CREATE VIEW                                 |
| Shutdown                 | Server Admin                          | To shut down the server                                            |
| Super                    | Server Admin                          | To use KILL thread, SET GLOBAL, CHANGE MASTER, etc.                |
| Trigger                  | Tables                                | To use triggers                                                    |
| Create tablespace        | Server Admin                          | To create/alter/drop tablespaces                                   |
| Update                   | Tables                                | To update existing rows                                            |
| Set user                 | Server                                | To create views and stored routines with a different definer       |
| Federated admin          | Server                                | To execute the CREATE SERVER, ALTER SERVER, DROP SERVER statements |
| Connection admin         | Server                                | To bypass connection limits and kill other users' connections      |
| Read_only admin          | Server                                | To perform write operations even if @@read_only=ON                 |
| Usage                    | Server Admin                          | No privileges - allow connect only                                 |
+--------------------------+---------------------------------------+--------------------------------------------------------------------+
41 rows in set (0.000 sec)

```

HB9HIL, 2024-09-27:
If the ajax for database migration fails there may be an issue with your webserver configuration. There are many reason why this can fail. 

Which webserver are you using?

pa8s, 2024-09-27:
I'm using nginx/1.26.1

HB9HIL, 2024-09-28:
> I'm using nginx/1.26.1 My wavelog instance is using roughly the same nginx config I use in my cloudlog instance.

we know about a bug in nginx. I created a seperate issue for that to track this down. 

Please refer any further issues in #1011

HB9HIL, 2024-09-28:
Closed as DUPE

Due the fact that it was mentioned multiple times I created a single Issue which tracks down the progress. 

See #1011

closed 2024-09-28 by HB9HIL

AndreasK79, 2024-09-28:
@pa8s did you change this https://github.com/wavelog/Wavelog/wiki/Installation#nginx-configuration in your nginx config before you tried to install Wavelog?

pa8s, 2024-09-28:
@AndreasK79 Yes, I did.

AndreasK79, 2024-09-28:
Ok. I had the migration crash during install when that wasn't correctly set, but I guess there are several reasons for it crashing. 

I could however access Wavelog and log in. I had to update contry files manually, but except for that, I could use the log.