wavelog/wavelog #1010: Wrong description for DL Gridmaster

kbabioch, 2024-09-27:
**Describe the bug**
When pressing the `Diplom Info` button on the `DL Gridmaster` page (`awards/gridmaster/dl`) in my wavelog instance (installed via Docker), I get a description for a different certificate (see screenshot below):

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS: Apache/2.4.62 (Debian)
- PHP:  8.3.11
- MySQL/MariaDB: 11.3.2
- Wavelog: 1.8.5

**Steps to Reproduce**
1. Go to `awards/gridmaster/dl`
2. Click on `Diplom Info`
3. Read the text and realize that it is not the one for `DL Gridmaster`

**Expected Behavior**
I expect the information for the `DL Gridmaster` certificate.

**Screenshots**

[image]

phl0, 2024-09-28:
There is no official DL Gridmaster. So it has the description for the US Gridmaster in there.

phl0, 2024-10-08:
Texts for non-US Gridmaster have been changed in https://github.com/wavelog/wavelog/pull/1019.