wavelog/wavelog #1011: Installation fails with nginx webserver

HB9HIL, 2024-09-28:
The Wavelog Installer fails with nginx webserver. 

It's basically the database migration which fails due to a infinite redirect which only appears in default nginx configs. 

Was already mentioned in #917

HB9HIL, 2024-09-28:
I was able to reproduce this issue. But currently I have no idea how to fix it. Maybe I have some time in the next two weeks to dig into this.

violarulan, 2024-09-29:
Tested with fresh install on my test machine and it works. The installer can finished by itself with all green.

I used rewrite config as follows

```
    # wavelog.conf
    # some static assets
    location /assets {
        try_files $uri $uri/;
    }
    # ....
    location / {
        try_files $uri $uri/ /index.php?$args;    # after installing, leave $config['index_page'] as empty in application/config.php
        include fastcgi_params;
        fastcgi_pass			unix:/run/php-fpm/php-fpm.sock;
        fastcgi_index			index.php;
        fastcgi_buffers			8 16k;
        fastcgi_buffer_size		32k;
        fastcgi_param DOCUMENT_ROOT	$realpath_root;
        fastcgi_param SCRIPT_FILENAME	$realpath_root$fastcgi_script_name;
    }    
```

HB9HIL, 2024-10-05:
Okay so probably not a bug and just a configuration issue. Thanks!

hellofinch, 2024-10-08:
Hello, this is my nginx's configuration file. Are there some wrong? I also leave $config['index_page'] as empty in application/config.php. But it still failed.
@violarulan 
```
server {
    listen 80 default_server;

    client_max_body_size 100M;

    root /app/public;
    index index.php index.html index.htm;

    location /pma/ {
        root /var/www/html;
        index index.php;

        location ~ \.php {
            fastcgi_buffer_size 128k;
            fastcgi_buffers 4 256k;
            fastcgi_busy_buffers_size 256k;

            fastcgi_pass unix:/run/php/php8.3-fpm.sock;
            fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name;
            fastcgi_param SCRIPT_NAME $fastcgi_script_name;
            include /etc/nginx/fastcgi_params;
            fastcgi_read_timeout 300;
        }
    }

    location /wavelog/ {
        root /var/www/html;
        # index index.php;
        try_files $uri $uri/ /index.php?$args;

        location ~ \.php {
            fastcgi_buffer_size 128k;
            fastcgi_buffers 4 256k;
            fastcgi_busy_buffers_size 256k;

            fastcgi_pass unix:/run/php/php8.3-fpm.sock;
            fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name;
            fastcgi_param SCRIPT_NAME $fastcgi_script_name;
            include /etc/nginx/fastcgi_params;
            fastcgi_read_timeout 3600;
        }
    }

    location / {
        try_files $uri $uri/ /index.php$is_args$args;
    }

    location ~ \.php {
        fastcgi_buffer_size 128k;
        fastcgi_buffers 4 256k;
        fastcgi_busy_buffers_size 256k;

        fastcgi_pass unix:/run/php/php8.3-fpm.sock;
        fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name;
        fastcgi_param SCRIPT_NAME $fastcgi_script_name;
        include /etc/nginx/fastcgi_params;
        fastcgi_read_timeout 300;
    }

    location ~ /\.ht {
        deny all;
    }
}
```
Server Software nginx/1.18.0
MariaDB 10.6.18-MariaDB-0ubuntu0.22.04.1 - Ubuntu 22.04
PHP 8.3.6 (cli) (built: Apr 11 2024 20:23:38) (NTS)

hellofinch, 2024-10-08:
THANKS!!! I SUCCESS! I modified the `try_file` path and leave `config['index_page']` blank. It works.
@HB9HIL Maybe it is a bug. After installation, `config['index_page']` in application/config/config.php should be blank manually.

HB9HIL, 2024-10-08:
thx will check later

HB9HIL, 2024-10-11:
@hellofinch and @violarulan 

I did some research and built this configuration which works for Wavelog quite good. 

```
server {
	listen 80 default_server;
	listen [::]:80 default_server;

	root /var/www/wavelog;

	# Add index.php to the list if you are using PHP
	index index.php index.html;

	server_name _;

    # index.php fallback
    location / {
        try_files $uri $uri/ /index.php?$query_string;
    }

    # favicon.ico
    location = /favicon.ico {
        log_not_found off;
    }

    # robots.txt
    location = /robots.txt {
        log_not_found off;
    }

    # assets
    location ~* \.(?:css(\.map)?|js(\.map)?|jpe?g|png|gif|ico|cur|heic|webp|tiff?|mp3|m4a|aac|ogg|midi?|wav|mp4|mov|webm|mpe?g|avi|ogv|flv|wmv)$ {
        expires 7d;
    }
    
    # svg, fonts
    location ~* \.(?:svgz?|ttf|ttc|otf|eot|woff2?)$ {
        add_header Access-Control-Allow-Origin "*";
        expires    7d;
    }

    # gzip (optional)
    # gzip            on;
    # gzip_vary       on;
    # gzip_proxied    any;
    # gzip_comp_level 6;
    # gzip_types      text/plain text/css text/xml application/json application/javascript application/rss+xml application/atom+xml image/svg+xml;

    # handle .php
    location ~ \.php$ {
        fastcgi_pass unix:/var/run/php/php-fpm.sock;

        # 404
        try_files                     $fastcgi_script_name =404;

        # default fastcgi_params
        include                       fastcgi_params;

        # fastcgi settings
        fastcgi_index                 index.php;
        fastcgi_buffers               8 16k;
        fastcgi_buffer_size           32k;

        # fastcgi params
        fastcgi_param DOCUMENT_ROOT   $realpath_root;
        fastcgi_param SCRIPT_FILENAME $realpath_root$fastcgi_script_name;
    } 
}
```

I tested this in my PR #1072 as I fixed a bug there which affects the database migration in the installer.

HB9HIL, 2024-10-11:
IMHO we can place this one in the Wiki

hellofinch, 2024-10-12:
WOW!!!
I tested in my device modified from your configuration. It works fine!
And some issues disappeared like [this](https://github.com/wavelog/wavelog/discussions/1059).
THANKS!!
I think we can place this in the wiki for anyone who uses Nginx.
73 :)

HB9HIL, 2024-10-12:
I already did :) 

https://github.com/wavelog/wavelog/wiki/Webserver-Configurations

HB9HIL, 2024-10-27:
Still not sure if this is a bug or not.. Let's keep this open

HB9HIL, 2024-10-27:
Issue #1132 was moved to discussions:

https://github.com/wavelog/wavelog/discussions/1133

HB9HIL, 2024-10-27:
I'm no nginx expert. So if an experienced nginx user could check would be great.

We start here: https://github.com/wavelog/wavelog/wiki/Webserver-Configurations and here https://github.com/wavelog/wavelog/discussions/1133

fabianfrz, 2025-01-01:
I installed a test instance using nginx / php-fpm 8.3.x / Manjaro / MariaDB.

Worked flawlessly with the config file provided in the wiki.

HB9HIL, 2025-01-01:
@fabianfrz 

yeah I guess we can close this one :) thx for feedback

closed 2025-01-01 by HB9HIL