wavelog/wavelog #1022: Backup of all user QSOs as ADIF

Loki1737, 2024-09-30:
**What Feature is Missing?**
Hi,
 I am missing a feature, to perform a backup of all QSOs as an ADIF via the backup operation as an admin. Currently, only Admins do have the option to perform a backup, but they can only see their own QSOs in the backup file.

**Why is This Feature Important?**
As a responsible admin, it would be nice, if this adif backup can be automated, to enable a restoration of QSOs.
Also adding it to the cron page would also be a huge benefit, for automation of this task.

**Is This Feature Personal or Beneficial to Others as Well?**
I think, it is beneficial for everyone.

int2001, 2024-10-01:
Hi @Loki1737 - tbh: this won't help, since station_profile Informations and settings aren't part of the ADIF.

Suggested way to backup your instance is a simple sql-dump of the whole database.
Have a look here: https://www.sqlshack.com/how-to-backup-and-restore-mysql-databases-using-the-mysqldump-command/

To Backup the produced files afterwards use a backup-solution, like 
- duplicity
- borg
- ...

Wavelog is no operating System. We follow the linux-rule: "Do one job, and do this one good". So why reinvent the wheel, if there're already tons of tools supporting the requested feature?

phl0, 2024-10-04:
I agree

HB9HIL, 2024-10-05:
I think we could deprecate the backup feature in the admin menu, BUT we could replace this with some kind of "User-based Backup" which not only contains all QSOs, but also Settings, Locations, Logbooks, and so on. 

Just an idea which would require a lot of work. So nothing which will be implemented real quick. I guess @phl0 already started something in this direction with #969

AndreasK79, 2024-10-06:
I don't use it, but it doesn't mean others are not using it. If we replace it with something else, sure. This can be used with cron as well, as API-key support was built in for ADIF and notes a while ago.

closed 2024-10-15 by github-actions[bot]