wavelog/wavelog #1030: LoTW Tooltip shows up twice - Frontend Bug

HB9HIL, 2024-10-01:
I’ve noticed a small issue that occurs when I start logging a QSO (which I often do just to gather information about a callsign without actually completing the log) and leave the browser tab open. When I later begin logging a QSO for a different callsign and hover the mouse pointer over the LoTW image, I sometimes see two black popups, or occasionally just one, showing LoTW data from the previous callsign I had started logging but didn’t finish. It seems that the wrong LoTW image is also displayed in these cases.

Thank you for everything you've done so far! This is just a small observation that I hope can help improve the experience even further.

[image]

Browser used Mozilla Firefox 130.0.1 (64-bit) on Windoze

_Originally posted by @besynnerlig in https://github.com/wavelog/wavelog/issues/978#issuecomment-2386844417_

HB9HIL, 2024-10-05:
I was not able to reproduce this :( it's difficult if it happens just "sometimes" and just in Firefox..

besynnerlig, 2024-10-06:
I can share a video showing it. Anyway, this doesn't bother me a lot. It might not be worth investigating, I don't know. 

https://github.com/user-attachments/assets/9ff1fc41-ed5a-45c5-93d4-9042ad062ecf

phl0, 2024-10-06:
Cannot reproduce either ...

HB9HIL, 2024-10-06:
[image]
got it

investigating

HB9HIL, 2024-10-06:
[image]

for documentation. will investigate later this day

HB9HIL, 2024-10-07:
fixed

PR #1049 (2024-10-07): LoTW Tooltip Bug in QSO logging
There was a little frontend bug in the QSO logging which only seems to appear in Firefox Browsers.

[image]

This fixes this and also adds translations for this text.

`Fixes #1030`
closed 2024-10-07 by HB9HIL