wavelog/wavelog #1046: Advanced Logbook - Globe Map qso path no correct with far side of earth station

wafer-li, 2024-10-06:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
If you contact a station located at the far side of the earth, the Globe map will show its path line straight through inside of the sphere.

Showing as the following image, the red line seems straight through the earth sphere at the yellow area.

[image]

**Expected Behavior**
The QSO Path should be visible in all its path, just like the qso with the near side of earth station.

violarulan, 2024-10-06:
#1048 should brings a fix.

wafer-li, 2024-10-09:
Upgraded to 1.8.6, maybe 0.35 still not enough

[image]

You could see the sink area shrink, but still below the horizon

wafer-li, 2024-10-09:
> Upgraded to 1.8.6, maybe 0.35 still not enough
> 
> <img alt="image" width="822" src="https://private-user-images.githubusercontent.com/12459199/374820480-82bed282-b5b2-4ca6-8b26-aec52a15d0f4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg0NTMxNzIsIm5iZiI6MTcyODQ1Mjg3MiwicGF0aCI6Ii8xMjQ1OTE5OS8zNzQ4MjA0ODAtODJiZWQyODItYjViMi00Y2E2LThiMjYtYWVjNTJhMTVkMGY0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA5VDA1NDc1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE5ODg1N2EyNmQwN2ZkNGUwOTU0ODlhNzI0NmE1ZjU3MDA2MmI1ZWUxZjc2MDBlMmNlNjRlNjE1Mjc3MTQyZDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.96_kHlox-FmYxRpSlMVg9Bn4UDxKAqCCyduBlFEyf7o">
> You could see the sink area shrink, but still below the horizon

Provide my case to be verified.

BI1QJQ OM75
CE3SOC FF46

HB9HIL, 2024-10-09:
fixed directly in `dev` branch

HB9HIL, 2024-10-09:
[image]

with value `37` instead `35` it's just fine

wafer-li, 2024-10-14:
> <img alt="image" width="743" src="https://private-user-images.githubusercontent.com/80885850/374842620-83016e02-e0d3-4081-8232-b6d7419b2cff.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg5MTI1MjAsIm5iZiI6MTcyODkxMjIyMCwicGF0aCI6Ii84MDg4NTg1MC8zNzQ4NDI2MjAtODMwMTZlMDItZTBkMy00MDgxLTgyMzItYjZkNzQxOWIyY2ZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE0VDEzMjM0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3MjdlYjFmMTNlODVlZDFhMWI3NzMwYzcxZGI5YTA4ZmJjNjRkYTAxMTBkYzRiMTgyYzhjZTc5M2FmNGVhYWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.g73LJxQC9sh-ym614o3kuxFRnEQ2SQUkubctOB6Mbd8">
> with value `37` instead `35` it's just fine

can't wait for next release

wafer-li, 2024-10-21:
After 1.8.7, I confirm this problem is totally resolved. @HB9HIL

closed 2024-10-21 by wafer-li