wavelog/wavelog #1055: Fix certain QSOs are not shown in gridmap

violarulan, 2024-10-08:
This is another part of #992 which I forgot to test.

We fixed that some unnecessary QSOs were shown in 2-digit grid locator condition by only appending the `%` to the end of the query.

Now if we have a QSO whose GL are more than one, for example `PN17,PN27,PN16,PN26` (right at 4 grids intersection), we can only see the QSO in the pop out list when clicking the `PN17` grid, nothing shown when clicking `PN27` or `PN16` or `PN26`.

Will try to submit a PR later this day.

phl0, 2024-10-08:
I can at least confirm the bug with a QSO on the grid intersection of four grids. Will test the patch.

phl0, 2024-10-08:
Fix merged to dev in https://github.com/wavelog/wavelog/pull/1056.

closed 2024-10-08 by violarulan