wavelog/wavelog #1058: Starttime doesn´t stop when leaving the input field for the call

OE3KSS, 2024-10-08:
**Describe the bug**
If you decide to log only the starttime of a qso (set at the wavelog-settings in the user account), the time doesn´t stop when leaving the input field for the call. Therefore the time of the end of the qso is logged when you finish with the enter-key.
When the system is set to store both times, everything is OK.

**Version Information**
- OS:  Apache/2.4.62 (Debian)
- PHP:     8.3.12
- MySQL/MariaDB:    11.3.2-MariaDB-1:11.3.2+maria~ubu2204
- Wavelog:      1.6.8 

**Steps to Reproduce**
1. Go to Useraccount - wavelog settings - logging options - log endtime seperatly -> OFF
2. Go to QSO - Live QSO
3. Put a Callsign to the input field and leave it with the tab-key
4. See error: The starting time doesn´t stop. 
5. When you store the QSO using the enter-key the current (end-)time is stored as starttime 

**Expected Behavior**
The time should stop when the input field for the call is left and stored as the starting time when the QSO is saved.

**Screenshots**
Include screenshots if applicable to illustrate the issue.

**Desktop Information:**
- OS:              Windows 10
- Browser:      Firefox 
- Version:       128.3.0esr

**Smartphone Information:**
- Device:       Motorola G84 5G
- OS:             Android 14
- Browser:     Chrome
- Version:      129.0.6668.81

**Additional Context**
Provide any other relevant context about the problem.

phl0, 2024-10-08:
Well this is kind of a philosophical question... It  was the original code IIRC to store the time upon logging the QSO (i.e. not the start time). This was not changed by adding the end time feature. So to be honest this is a 50/50 thingy imho. Will discuss in dev team.

phl0, 2024-10-08:
Had a short discussion re this issue and we came to the conclusion to leave it as it is. Your argument is valid but on the other hand we have people working in pile-ups who enter the call and tab to the RST fields waiting for their call. Those would want the time logged when they press enter. So if you are interestend in the start time enable the separate logging of end time and you will have both.

closed 2024-10-10 by phl0