wavelog/wavelog #1066: API/QSO - Import Count Mismatch: Successful Response Despite No QSO Import

besynnerlig, 2024-10-10:
I am using the API/QSO endpoint as described in the [Wavelog Wiki](https://github.com/wavelog/wavelog/wiki/API#apiqso) for logging QSOs from a third-party app into Wavelog.

Following the example provided in the Wiki, I use the command below:

```bash
curl -X POST https://<WAVELOG_URL>/index.php/api/qso -H "Content-Type: application/json" -H "Accept: application/json" -d "{"key":"<API_KEY>","station_profile_id":"<STATION_PROFILE_ID>","type":"adif","string":"<call:5>N9EAT<band:4>70cm<mode:3>SSB<freq:10>432.166976<qso_date:8>20190616<time_on:6>170600<time_off:6>170600<rst_rcvd:2>59<rst_sent:2>55<qsl_rcvd:1>N<qsl_sent:1>N<country:24>United States Of America<gridsquare:4>EN42<sat_mode:3>U/V<sat_name:4>AO-7<prop_mode:3>SAT<name:5>Marty<eor>"}"
```

This command works perfectly, and the response is:

```json
{"status":"created","type":"adif","string":"","imported_count":1,"messages":[""]}
```

However, if I attempt to import a QSO where the station callsign in the ADI string does not match the active station profile (for example, `<station_callsign:4>SE6S`), the response from the API remains identical:

```json
{"status":"created","type":"adif","string":"","imported_count":1,"messages":[""]}
```

Even though the QSO does not import successfully due to the callsign mismatch, the `imported_count` field still shows a count of 1.

**Expected Behavior**  
I expected the `imported_count` field to reflect the actual number of successfully imported QSOs (in this case, it should be 0). An alternative approach could be to add a new field, such as `successful_imported_count`, to clarify how many QSOs were truly imported, allowing third-party apps to better handle these discrepancies.

**Version Information**  
- Wavelog: 1.8.6

**Steps to Reproduce**  
1. Use the API/QSO endpoint to submit an ADIF string with a station callsign that doesn't match the active station profile.  
2. Review the JSON response and observe that `imported_count` is 1, even though no QSOs were actually imported.

**Expected Behavior**  
The `imported_count` field should reflect 0 when a QSO fails to import due to a station callsign mismatch.

**Additional Context**  
This small adjustment would help third-party apps accurately determine whether QSOs were successfully imported, reducing potential confusion or incorrect data handling. While I understand that the documentation in the Wiki doesn't explicitly describe the structure of the JSON response from the API, my assumption about the behavior is based on the field names in the response.

phl0, 2024-10-10:
Fix proposed in https://github.com/wavelog/wavelog/pull/1067. Please test.

besynnerlig, 2024-10-10:
Tested and works! Thanks for the PR!

closed 2024-10-10 by besynnerlig