wavelog/wavelog #1077: Inconsistent cron job status indication

corvus-ch, 2024-10-13:
Cron status are indicated differently depending on if the page is loaded via URL or if using the refresh button.

**Version Information**
- OS: Apache/2.4.62 (Debian)
- PHP:  	8.3.12
- MySQL/MariaDB: 11.5.2-MariaDB-ubu2404
- Wavelog: 1.8.6
- Docker Image: ghcr.io/wavelog/wavelog@e10139c03457 

**Steps to Reproduce**

_Case 1_

1. Visit `/cron`
2. See failed jobs with correct translation highlighted in red
3. Press "Aktualisiere Liste"
2. See failed jobs with English text highlighted in yellow.

_Case 2_

1. Visit `/cron`
2. See failed jobs with correct translation highlighted in red
3. Edit a job
2. See failed jobs with English text highlighted in yellow.

_Case 3_

1. Visit `/cron`
2. See failed jobs with correct translation highlighted in red
3. Enable or disable a job
2. See failed jobs with English text highlighted in yellow.

**Expected Behavior**

* Language does not switch
* Consistent status colors

**Screenshots**

_On initial load_

[image]

_after refresh_

[image]

**Desktop Information:**

- OS: macOS 14.6.1 (23G93)
- Browser: Firefox
- Version: 131.0.2 (64-Bit)

**Additional Context**

Relates to https://github.com/wavelog/wavelog/discussions/1076

HB9HIL, 2024-11-05:
@corvus-ch 

Should be fixed in `dev` branch :)

corvus-ch, 2024-11-10:
What is the best way for me to test and verify this? It is probably a bad idea to just update my production instance. Is there even a container image available for the dev branch?

HB9HIL, 2024-11-11:
> What is the best way for me to test and verify this? It is probably a bad idea to just update my production instance. Is there even a container image available for the dev branch?

There is a docker container available for the `dev` instance :) instead `latest` just choose `dev`. See our Docker Wiki for more informations ([here](https://github.com/wavelog/wavelog/wiki/Installation-via-Docker))

To test this there are multiple ways. To verify the fix set up a local instance of Wavelog and play around. You could prohibit internet access to this container (the cron runs against localhost in docker) and see if the jobs fail. Also set the instance to german and see if the text is still german after hitting the reload button.

HB9HIL, 2024-11-11:
probably this month