wavelog/wavelog #1080: button style wrong

hellofinch, 2024-10-14:
**This is for reporting bugs with the code, if this issue is regarding setting up on your local server please use the discussions area**

**Describe the bug**
In mode page, the button clicked but the color doesn't change.
And the translation missed.

[image]

AndreasK79, 2024-10-14:
Fix here https://github.com/wavelog/wavelog/pull/1083.

Thanks for reporting.

HB9HIL, 2024-10-15:
@hellofinch 

I added the translatables to the PR aswell. After merge the strings are translatable in weblate -> https://translate.wavelog.org

hellofinch, 2024-10-16:
I believe this is a display issue, and there is no problem with the translation.
I see something not translated into Chinese. I will contribute some translations in weblate. :)

HB9HIL, 2024-10-16:
> I believe this is a display issue, and there is no problem with the translation. I see something not translated into Chinese. I will contribute some translations in weblate. :)

some of the javascript things weren't translatable yet. this is fixed now. 

small hint: weblate is offline from now on for approx. 1 hour (16th oct 2024 0700z to 0800z)

closed 2024-10-21 by hellofinch