wavelog/wavelog #1102: Edit a QSO within the live contest logging window

winnieXY, 2024-10-20:
**What Feature is Missing?**
Within the live contest logging window there is a qso list below. If you made a error (e.g. got the callsign wrong) you are currently not able to edit that qso within that window (but have to open another one). 

I would like to have the callsign clickable just as on the frontpage in order to be able to open and edit the qso. 

**Why is This Feature Important?**
Easing the contest handling within wavelog.

**Is This Feature Personal or Beneficial to Others as Well?**
I do think so - everybody does make errors and it would be nice if you can correct these as easy as possible :-)

205er, 2024-10-21:
That would be sooo helpful!

AndreasK79, 2024-10-21:
I added the code here #1103

closed 2024-10-30 by github-actions[bot]