wavelog/wavelog #1109: Possibility to add multiple operators to a QSO

m0urs, 2024-10-22:
**What Feature is Missing?**
I am missing a possibility to add more than one operator if I am logging a QSO. E.g. I would like to add 

"CALL1,CALL2"

in the "Callsign" field and if I then submit the QSO, two QSOs will be created where all fields but the callsign will be identical.

**Why is This Feature Important?**
This is very helpful if you have e.g. a POTA contact with multiple operators but also if you talk to a club station and you would like to log the QSO also with the personal callsign of the operator. Currently you need to manually copy all the fields and log an additional QSO.

**Is This Feature Personal or Beneficial to Others as Well?**
From my opinion this would be helpful also for others.

dg9vh, 2024-10-22:
In my opinion this idea is good, but logging should result in separate QSOs. Means following workflow:

Entering multiple calls separated with comma.
Pushing Save creates a number of logged QSOs with same time, sate, qrg, mode but without personal data.  

These could be added later by hand or by update from callbook.

m0urs, 2024-10-22:
Yes, that is exactly what I meant. However, the only field which does not need to be copied  is the "Name" I guess?

HB9HIL, 2024-10-23:
We discussed a little bit about that in the dev team. The team and myself is not sure about this feature request. 

> This is very helpful if you have e.g. a POTA contact with multiple operators

This might be a help and a little bit faster, but still it means some logical discussions about which fields should be populated to the QSO data and which not. A lot of effort and code work for a real edge case.

> club station and you would like to log the QSO also with the personal callsign of the operator

This would be in fact bad practice. If you have a QSO with a club callsign you do NOT have a QSO with the operator himself. This person is just the operator. Although I can understand the need for that, there is no offical ADIF field for the **contacted stations operator callsign**, but only for the name which is our "Name" field. I don't think we should support that as it would create a (per definition) invalid QSO. Sure it can be discussed, but - and this is my personal opinion - this is something I would not implement.

ethancedwards8, 2025-01-01:
> This would be in fact bad practice. If you have a QSO with a club callsign you do NOT have a QSO with the operator himself. This person is just the operator. Although I can understand the need for [...]

That's actually not true, at least for POTA: https://docs.pota.app/docs/activator_reference/activator_guide_clubs.html#how-activation-credit-is-awarded

HB9HIL, 2025-01-01:
> That's actually not true, at least for POTA: https://docs.pota.app/docs/activator_reference/activator_guide_clubs.html#how-activation-credit-is-awarded

This has nothing to do with the issue here. 

POTA says about the adif which was uploaded by the clubstation:
You get credits for the station callsign and individual credits for the operators callsign.

This is totally fine as ADIF has a field called `OPERATOR` which holds the callsign of the operator of the current station callsign. 

--- 

You have to keep in mind that for one QSO alsways two ADIF files exists (assuming both stations are logging properly). If you log a QSO with Station A which is operated by Operator 1 then you can only log the Callsign of  Station A. At Station A they log your callsign as `station_callsign` and as operator your personal callsign. 

There is - as already mentioned - NO adif field which hold the operator callsign of the OTHER station. Only the operator of your own station is documented in the ADIF field. And for one QSO there are always two adif files - assuming both stations are logging correctly.

AndreasK79, 2025-05-27:
As this is something that will not be implemented, it is being closed.