wavelog/wavelog #1116: Can't log QSO with specific callsign

xyz667, 2024-10-22:
I wanted to log a QSO with TI5VMJ today, however it was impossible to save this QSO.

**Version Information**
We need some information about your versions. You can get all Information in the Admin > Debug Menu
- OS:               DietPi 9.7.1 (Bookworm)
- PHP:              8.2.24-1
- MariaDB:      1:10.11.6-0
- Wavelog:       1.8.8

**Steps to Reproduce**
1. Go to Live QSO or Post QSO
2. Enter callsign TI5VMJ and leave the callsign field so QRZ data is populated into name location and gridsquare fields (valid data seems to be filled in the fields)
3. Try to save QSO
4. Save button does nothing, QSO can not be saved. No error messages.

**Expected Behavior**
Let me save.

**Desktop Information:**
- OS:               macOS
- Browser:          Safari
- Version:          15.0.1

**Smartphone Information:**
- Device:           different iPhone
- OS:               18.0 and 18.1
- Browser:          Safari

xyz667, 2024-10-22:
> Hello, the problem with this specific callsign is, that QRZ.com does not deliver a CQ-Zone. You should add a correct CQ-Zone by hand, then it would be logged.

But why is that a problem? If I don’t leave the callsign field so that no QRZ data is fetched I can save without any problem (and without CQ zone I guess)?
I should be able to leave any fields empty, or get some sort of hint/error should they be mandatory.

HB9HIL, 2024-10-22:
Hello @xyz667 

I can't reproduce your issue. The CQ Zone as @dg9vh shouldn't be an issue. 

Can you please the the followings:

- Make sure you updated your DXCC data, in the best case you've got the cronmanager running and update dxcc data repeatly

- Pleas enable error logging. For this you need to edit your `application/config/config.php`. Look for `$config['log_threshold'] = 0` and set this value to `1`. Then reproduce the issue and check the logfile which is located in `application/logs/log-[date].php`. Let us know if you see any ERROR's.

- Pleas update to the latest version which is `1.8.8` which was released today. We had a bug in the QSO logging, maybe it's related to that. 

Thanks for your feedback

HB9HIL, 2024-10-22:
@xyz667  and @dg9vh 

I'm Sorry @dg9vh. You are probably right. Seems we have an issue here with the CQ Zone. I'm gonna fix that

HB9HIL, 2024-10-22:
@xyz667 
to temporarily solve that just choose a CQ zone before hitting the Save button. For TI5VMJ it's CQ Zone 7 😉

xyz667, 2024-10-23:
> @xyz667 to temporarily solve that just choose a CQ zone before hitting the Save button. For TI5VMJ it's CQ Zone 7 😉

Thanks a lot! Still there should be some sort of error message to the user, right?

closed 2024-10-29 by xyz667