wavelog/wavelog #1119: Show eqsl status in popup window (displayContacts) when clicking on today received eqsls

winnieXY, 2024-10-23:
**What Feature is Missing?**
When clicking on  todays received eqsls another window opens which displays the today confirmed qsos:

[image]

However, within this window it's impossible to see the eqsl status or view the received cards as the eqsl row is missing:
[image]

(Obviously I got no confirmed eqsls today so far ;-))

**Why is This Feature Important?**
Important is the wrong word - nice would be the better one. :-) to get a short overview over the received cards. 

**Is This Feature Personal or Beneficial to Others as Well?**
I guess this will be beneficial also for others.

xyz667, 2024-10-23:
eQSL is showing as a table header for me in that popup (but no Clublog as I don't use it).
Can't confirm content of that column as I don't have confirmations today either. ;-)

int2001, 2024-10-23:
i don't get the point here.
if there's no confirmation via eQSL "today", the popup-table is empty.

we can disable the link, if the eQSL-Counter is Zero? Is this the issue?!

int2001, 2024-10-23:
PS: We discussed internally a bit with an enabled "glass ball".
Do you mean the COLUMN eQSL which ist missing IF there are eQSL's today?

if: pse go to settings and add eQSL as a default QSL-Method.

winnieXY, 2024-10-23:
@int2001 : sorry for being unclear. Yes. This was the intented meaning. Thanks for the hint.

closed 2024-10-23 by winnieXY