wavelog/wavelog #1125: Advanced Logbook: "QSL Filter" ignores "Options for the Advanced Logbook"

xyz667, 2024-10-24:
In the "Options for the Advanced Logbook" (Options button) I can disable columns to be displayed, e.g. for QSL services I do not use.
But when I disable "Clublog" in the Options, the QSL Filter still has Clublog selections, needlessly cluttering the already quite overloaded display.

Version Information

OS: DietPi 9.7.1 (Bookworm)
PHP: 8.2.24
MySQL/MariaDB: 10.11.6-MariaDB-0+deb12u1
Wavelog: 1.8.8

Steps to Reproduce
Open Advanced Logbook, disable Clublog in the Options for the Advanced Logbook and save.
Now open the QSL-Filter.
There are still selections for Clublog sent and received, although there are no Clublog columns in the logbook view.

Expected Behavior
a) Columns I disable should be gone anywhere.
b) Better alternative IMO (although you design choice seems to be different): If there are no userid+password for a QSL service it should be hidden everywhere in Wavelog.

Desktop Information:

OS: macOS 15.0.1
Browser: Safari

AndreasK79, 2024-10-24:
Yeah, cluttering is always a problem. I am open to other ways of doing it. Cluttering is also why I implemented buttons to show/hide filters. I will implement hiding of filters when columns are not shown. I already did that for quickfilters, but not the other filters.

AndreasK79, 2024-10-31:
The improvements are in the dev branch. If you disable columns, the filters are now hidden.

closed 2024-11-08 by github-actions[bot]